platformio/platformio-core #786: Compilation fails - can't resolve library headers for linux_arm

piotrbazan, 2016-09-17:
### Configuration

**Operating system**: ubuntu 16.04, arm

**PlatformIO Version** (`platformio --version`):PlatformIO, version 3.0.1

### Description of problem
When project has dependency on a library compilation for arduino works ok, however for raspberry pi there is a compilation error:  fatal error: DallasTemperature.h: No such file or directory

#### Steps to Reproduce
1. Create project with:
```ini
[env:uno]
platform = atmelavr
framework = arduino
board = uno
libs = 
  54
[env:raspberrypi_zero]
platform = linux_arm
framework = wiringpi
board = raspberrypi_zero
libs =
  54
```
2. issue: pio run

### Actual Results
```
Dependencies found For uno:
Looking for dependencies...
Library Dependency Graph
|-- <DallasTemperature> v3.7.7
|   |-- <OneWire>

For respberry:
Looking for dependencies...
Project does not have dependencies
Compiling .pioenvs/raspberrypi_zero/src/main.o
src/main.cpp:1:31: fatal error: DallasTemperature.h: No such file or directory
```
### Expected Results
Dependencies found and compilation works.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:uno]
platform = atmelavr
framework = arduino
board = uno
libs = 
  54
[env:raspberrypi_zero]
platform = linux_arm
framework = wiringpi
board = raspberrypi_zero
libs =
  54
```

**Source file to reproduce issue:**
```cpp
#include <DallasTemperature.h>
int main(void) {}
```

### Additional info

ivankravets, 2016-09-17:
1. What does mean ``libs = 54``? Do you mean http://docs.platformio.org/en/stable/projectconf.html#lib-deps ?
2. Run in verbose mode `pio run -v` and you will receive an answer. 
3. This option could be interested for you http://docs.platformio.org/en/stable/librarymanager/ldf.html#ldf-compat-mode

ivankravets, 2016-09-17:
I've just taken a look at your fork https://github.com/piotrbazan/platform-linux_arm. Cool work!

How about to have 1 development platform for both boards: RPi + Odroid? We can have 2 "cores" in `framework-wiringpi` package. We have the same for Arduino and other frameworks. Then in build script we select apropriate core for building.

I think that your work could be interested for other developers. Also, we would like to see PR to main development platform.

ivankravets, 2016-10-26:
Do you have any news on this issue?

ivankravets, 2016-12-05:
DallasTemperature library is compatible with Arduino framework and PlatformIO checks it. You are going to use other framework named "wiringpi". In this case, you need turn off LDF compatibility mode. See http://docs.platformio.org/en/stable/projectconf.html#projectconf-lib-compat-mode

```ini
[env:uno]
platform = atmelavr
framework = arduino
board = uno
lib_deps =
  54

[env:raspberrypi_zero]
platform = linux_arm
framework = wiringpi
board = raspberrypi_zero
lib_deps =
  54
lib_compat_mode = 0
```

closed 2016-12-05 by ivankravets