platformio/platformio-core #787: Feature Request: Configure "platformio device monitor" from platformio.ini

blackketter, 2016-09-18:
It should be possible to configure “platformio monitor” options from platformio.ini
 
I'd love to be able to add a couple lines to my platformio.ini file like:

monitor_port = /dev/cu.wchusbserial1424120
monitor_speed = 9600

and this would configure "platformio device monitor" on a per-target basis.  

Ideally, all the "device monitor" options would be available this way.

Bonus idea:  "platformio run -t monitor" would monitor, similarly to uploading via "platformio run -t upload"

blackketter, 2016-09-18:
To elaborate on the last idea, I think it would be awesome to be able to do something like this:

`platformio run -e nodemcu -t clean -t upload -t monitor`

Which would clean, build, upload and then monitor the nodemcu environment.   (Maybe this should be a separate feature request?)

ivankravets, 2016-09-18:
`pio run` is responsible for building and uploading. I'm not sure that we should mix `pio device` commands with `pio run`. If you need that, you can create alias in OS.
```shell
alias piobatch="pio -t clean && pio -t upload && pio device monitor"
```

blackketter, 2016-09-18:
Fair enough, I already have bunch of similar aliases. :)  

I figured it would be generally useful for folks to be able to do this kind of build/install/monitor cycle within a single command and it wasn't obvious to me why `pio run` uploads but doesn't monitor.

I guess then, we'd need to have a -e option on `pio device monitor` so it knew which environment to pull the monitor options from in platformio.ini. 

So the with an environment-aware device monitor the alias would be:

`alias pionodebatch="pio -e nodemcu -t clean && pio -e nodemcu -t upload && pio -e nodemcu device monitor "`

ivankravets, 2016-09-18:
Good, please open a separate issue. I think we will find solution for that 😊 

P.S: How is PlatformIO 3.0? Do you have problems with it?

blackketter, 2016-09-18:
PlatformIO 3.0 seems great so far, no problems!  Great job!

ivankravets, 2016-09-18:
Thanks!!! PlatformIO 3.1 is coming this week! :) 
* https://github.com/platformio/platformio/blob/develop/HISTORY.rst

ivankravets, 2016-09-18:
Dynamic variables http://docs.platformio.org/en/latest/projectconf.html#dynamic-variables

Also, +50 new boards :)

blackketter, 2016-09-18:
SWEET!

ivankravets, 2017-06-06:
Please `pio upgrade` to b11

http://docs.platformio.org/en/latest/projectconf.html#monitor-options

commit b04fc327 (2017-06-06): Configure Serial Port Monitor from `platformio.ini` // Resolve #787
closed 2017-06-06 by ivankravets