platformio/platformio-core #795: integer arguments in for build_flags are missing

Palatis, 2016-09-26:
I'm starting a new platform, and am trying to write my `board.json`.

I observed that arguments starts with `[0-9]+` are missing in pio action command line.
for example,
```
{
  "build": [
    "extra_flags": "-DN79E85X -DN79E855 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_N79E85X --iram-size 256 --xram-size 256 --code-size 16384 --stack-size 128",
    ...
  ],
  ...
}
```
and `pio run -v` outputs something like:
```
sdcc -o .pioenvs/mcs51/src/ADC.rel -c --std-sdcc11 --opt-code-size --peep-asm --peep-return -mmcs51 --iram-size --xram-size --code-size --stack-size -DF_CPU=22118400L -DPLATFORMIO=030100 -DN79E85X -DN79E855 -DNAKED_ARCH_MCS51 -DNAKED_MCS51_N79E85X -I/home/palatis/.platformio/packages/framework-nakedmcs51/cores/naked -I/home/palatis/.platformio/packages/framework-nakedmcs51/variants/n79e85x -Isrc src/ADC.c
```
all the integer values from `build.extra_flags` is gone into the void.

and this doesn't work for `platformio.ini` either:
```
[env:myboard]
build_flags = --iram-size 256 --xram-size 256 --code-size 16384 --stack-size 128
```
same output from `pio run -v`

ivankravets, 2016-09-28:
Could you try `--iram-size=256 --xram-size=256?

Palatis, 2016-09-28:
I tried already, and sdcc doesn't understand that.

ivankravets, 2016-09-28:
That was your solution?

ivankravets, 2016-09-28:
The problem here that we use [SCons ParseFlags](http://scons.org/doc/production/HTML/scons-user.html). Search for "env.ParseFlags(".

I see only 1 solution here. Create a new field in board manifest, like "build.extra_flags_" and put here these flags. Then, within `main.py` in platform build script you will be able to put these flags where do you want. 
```
env.Append(
...
)
```

Palatis, 2016-09-28:
current solution is more like a "workaround".

I add
```
  "build": {
    "iram_size": 256,
    "xram_size": 256,
    "code_size": 4096,
    "heap_size": 128
  }
```
in `board.json` and parse them with `builder/main.py`.
https://github.com/Palatis/platform-mcs51/blob/master/builder/main.py#L98
https://github.com/Palatis/platform-mcs51/blob/master/boards/n79e855.json

ivankravets, 2016-09-28:
Yes, you are right. However, use 1 field and make array
```json
  "build": 
  {
      "extra_flags_": [
        "iram_size", "256",
        "xram_size", "256"
   ]
}
```

```python
env.Append(LINKFLAGS=env.BoardConfig().get("build.extra_flags_"))
```

Palatis, 2016-09-28:
will give it a try when i got time.
got other stuff to work on.

ivankravets, 2018-01-08:
We have a very easy solution for this issue. Please create PRE custom script http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

# pre:extra_script.py
```
Import("env")

env.Append(CCFLAGS=[
    "--data-loc", 0x30,
    " --disable-warning", 126,
    "--disable-warning", 59
])
```

zerog2k, 2018-01-08:
thanks @ivankravets 
fwiw, i was able to accomplish this by adding it to existing post extra_script, _and_ needed to separate the options into the CFLAGS (for compiler warnings) and LINKFLAGS (for the linker pass):

```
env.Append(
    CFLAGS = [
    "--disable-warning", 126,
    "--disable-warning", 59
    ],
    LINKFLAGS = [
        "--data-loc", 0x30
    ]
)
```

results in this from a pio run -v
```
post_extra_script running...
sdcc -o .pioenvs/stc15w408as/src/adc.rel -c --std-sdcc11 --disable-warning 126 --disable-warning 59 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200 -DHEAP_SIZE=128 -DPLATFORMIO=30501 -Isrc -Iinclude src/adc.c
sdcc -o .pioenvs/stc15w408as/src/ds1302.rel -c --std-sdcc11 --disable-warning 126 --disable-warning 59 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200 -DHEAP_SIZE=128 -DPLATFORMIO=30501 -Isrc -Iinclude src/ds1302.c
sdcc -o .pioenvs/stc15w408as/src/main.rel -c --std-sdcc11 --disable-warning 126 --disable-warning 59 --opt-code-size --peep-return -mmcs51 -DF_CPU=11059200 -DHEAP_SIZE=128 -DPLATFORMIO=30501 -Isrc -Iinclude src/main.c
sdcc -o .pioenvs/stc15w408as/firmware.ihx -mmcs51 --iram-size 256 --xram-size 256 --code-size 8096 --out-fmt-ihx --data-loc 48 .pioenvs/stc15w408as/src/adc.rel .pioenvs/stc15w408as/src/ds1302.rel .pioenvs/stc15w408as/src/main.rel -L.pioenvs/stc15w408as
```

ivankravets, 2018-08-20:
Resolved in https://github.com/platformio/platform-intel_mcs51/commit/fdd95046034ea0b62b54b4c088bde1a65aef142e

closed 2018-08-20 by ivankravets