platformio/platformio-core #801: ignored macros #ifdef

St007, 2016-10-08:
When I tried to compile project for  Digispark (Attiny85) it fails with errors about not defined SPI registers. It is true, Attiny85 does not have SPI, but somehow platformio decides took it and create folder SPI in .pioenvs

I found why this happens ( but it must not happen ! ) . The bug that there is some includes of SPI.h but they are not active (paced in #ifdef section). Platformio includes all that can be found also if it actually not included after processing macros. Temporary solved by manually deleting of all includes of SPI.h and only then project was compiled.
Here is defined RH_PLATFORM_ATTINY and no SPI definitions must be included, but platformio thinks in different way. (Forgive me I'm saying Platformio, for end-user it is like Platformio bug)

**RadioHead.h**
```cpp

#ifndef RadioHead_h
#define RadioHead_h

// Official version numbers are maintained automatically by Makefile:
#define RH_VERSION_MAJOR 1
#define RH_VERSION_MINOR 61

// Symbolic names for currently supported platform types
#define RH_PLATFORM_ARDUINO          1
#define RH_PLATFORM_MSP430           2
#define RH_PLATFORM_STM32            3
#define RH_PLATFORM_GENERIC_AVR8     4
#define RH_PLATFORM_UNO32            5
#define RH_PLATFORM_UNIX             6
#define RH_PLATFORM_STM32STD         7
#define RH_PLATFORM_STM32F4_HAL      8 
#define RH_PLATFORM_RASPI            9
#define RH_PLATFORM_NRF51            10
#define RH_PLATFORM_ESP8266          11
#define RH_PLATFORM_STM32F2          12
#define RH_PLATFORM_CHIPKIT_CORE     13

////////////////////////////////////////////////////
// Select platform automatically, if possible
#ifndef RH_PLATFORM
 #if (MPIDE>=150 && defined(ARDUINO))
  // Using ChipKIT Core on Arduino IDE
  #define RH_PLATFORM RH_PLATFORM_CHIPKIT_CORE
 #elif defined(MPIDE)
  // Uno32 under old MPIDE, which has been discontinued:
  #define RH_PLATFORM RH_PLATFORM_UNO32
 #elif defined(NRF51)
  #define RH_PLATFORM RH_PLATFORM_NRF51
 #elif defined(ESP8266)
  #define RH_PLATFORM RH_PLATFORM_ESP8266
 #elif defined(ARDUINO)
  #define RH_PLATFORM RH_PLATFORM_ARDUINO
 #elif defined(__MSP430G2452__) || defined(__MSP430G2553__)
  #define RH_PLATFORM RH_PLATFORM_MSP430
 #elif defined(MCU_STM32F103RE)
  #define RH_PLATFORM RH_PLATFORM_STM32
 #elif defined(STM32F2XX)
  #define RH_PLATFORM RH_PLATFORM_STM32F2
 #elif defined(USE_STDPERIPH_DRIVER)
  #define RH_PLATFORM RH_PLATFORM_STM32STD
 #elif defined(RASPBERRY_PI)
  #define RH_PLATFORM RH_PLATFORM_RASPI
#elif defined(__unix__) // Linux
  #define RH_PLATFORM RH_PLATFORM_UNIX
#elif defined(__APPLE__) // OSX
  #define RH_PLATFORM RH_PLATFORM_UNIX
 #else
  #error Platform not defined! 	
 #endif
#endif

#if defined(__AVR_ATtiny84__) || defined(__AVR_ATtiny85__) || defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtinyX4__) || defined(__AVR_ATtinyX5__) || defined(__AVR_ATtiny2313__) || defined(__AVR_ATtiny4313__) || defined(__AVR_ATtinyX313__)
 #define RH_PLATFORM_ATTINY
#endif

////////////////////////////////////////////////////
// Platform specific headers:
[... 218 lines cut ...]
  #define ntohs(x) htons(x)
  #define htonl(x) ( ((x)<<24 & 0xFF000000UL) | \
                   ((x)<< 8 & 0x00FF0000UL) | \
                   ((x)>> 8 & 0x0000FF00UL) | \
                   ((x)>>24 & 0x000000FFUL) )
  #define ntohl(x) htonl(x)

 #elif __BYTE_ORDER__ == __ORDER_BIG_ENDIAN__
  // Others
  #define htons(x) (x)
  #define ntohs(x) (x)
  #define htonl(x) (x)
  #define ntohl(x) (x)

 #else
  #error "Dont know how to define htons and friends for this processor" 
 #endif
#endif

// This is the address that indicates a broadcast
#define RH_BROADCAST_ADDRESS 0xff

#endif

```

ivankravets, 2016-10-08:
How do you set `RH_PLATFORM` macro? Could you try [lib_ignore](http://docs.platformio.org/en/stable/projectconf.html#projectconf-lib-ignore) option in `platformio.ini`?
```ini
[env:myenv]
platfrorm =..
lib_ignore = SPI
```

St007, 2016-10-08:
Hi, 
Thank you for quick replay.
In the file RadioHead.h (attached to my initial message) there several lines:
`#if defined(__AVR_ATtiny84__) || defined(__AVR_ATtiny85__) || defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtinyX4__) || defined(__AVR_ATtinyX5__) || defined(__AVR_ATtiny2313__) || defined(__AVR_ATtiny4313__) || defined(__AVR_ATtinyX313__)
 #define RH_PLATFORM_ATTINY
#endif`
and the platform.ini is as follow :
`[env:digispark-tiny]
platform = atmelavr
framework = arduino
board = digispark-tiny`

Adding lib_ignore = SPI solves the problem, but I still can't understand the source of the problem (I spent a lot of hours to figure the problem) and if it is safe to use Platformio for next projects.

ivankravets, 2016-10-08:
Could you define manually `RH_PLATFORM_ATTINY`?
```
[env:...]
build_flags = -DRH_PLATFORM_ATTINY
```

I'll explain later all details...

St007, 2016-10-08:
If I remove lib_ignore = SPI and add this
```
[env:digispark-tiny]
build_flags = -DRH_PLATFORM_ATTINY
```
compilation fails with same error as initially (not defined SPI). However RadioHead.h has warnings in case if RH_PLATFORM_ATTINY is defined and these warnings are visible, so everything is ok, RH_PLATFORM_ATTINY is defined.
```
 #ifdef RH_PLATFORM_ATTINY
  #warning Arduino TinyCore does not support hardware SPI. Use software SPI instead.
 #else
  #include <SPI.h>
  #define RH_HAVE_HARDWARE_SPI
  #define RH_HAVE_SERIAL
 #endif
```
But as I mentioned at the beginning, the SPI is included and compilation fails.

ivankravets, 2016-10-08:
Please try `pio run -t clean` and rebuild in verbose mode using `pio run -v`. Do you this macro in CLI?

Ok, let's em explain details. We introduced [Library Dependency Finder (LDF)](http://docs.platformio.org/en/stable/librarymanager/ldf.html) in PlatformIO 3.0. The role of this machinery is to automatically find project dependencies. LDF analyzes `#include` from source files and builds internal dependency map "on-the-fly". If you don't like it, you can turn-off this feature. See [Dependency Finder Mode](http://docs.platformio.org/en/stable/librarymanager/ldf.html#ldf-mode).

In 99% LDF works awesome!!! Super fast and reliable. Why is it cool? Because it doesn't depend on external GCC tools. We don't call `gcc -E` to fetch dependency for the one file. For example, Arduino IDE does it. This is expensive operation... Arduino IDE can use `gcc -E` because it doesn't provide project management and normal library/dependency manager. In the most case, newbies use 1 file(sketch) as a project... PlatformIO developers have big projects with dozens of files and dozens of library dependencies. Using `gcc -E` will increase build process in 10x time... Currently, PlatformIO Build System is No.1 in comparison with Arduino IDE, Make or CMake 3-rd party build systems for embedded. PlatformIO needs 1-3sec to build the whole project with the framework, where others need 15 sec and more for the same project. 

Of course, this super optimization has a side effect. As I've told before, we don't call `gcc -E` per each file and use Python implementation that interprets (emulates) C/C++ Preprocessor conditional syntax (#ifdef, if, defined, else, and elif). The issue here, that we don't know about internal(built-in)  macros in compiler. These macors such as `__AVR_ATtinyX313__` and etc are not visible for our Python-based build system. The solution here is to hint our build system about internal macros.

Finally, as LDF doesn't see `__AVR_ATtiny84__` and etc, the `RH_PLATFORM_ATTINY` is not defined on the pre-processing stage. That is why LDF sees  `#include <SPI.h>`.

P.S: Try to define some marcos in `platformio.ini` and check in project source files. In this case, LDF will work correctly.

St007, 2016-10-08:
Thank you very much, this clearly explains the behavior I was stuck with.

closed 2016-10-08 by St007