platformio/platformio-core #802: Build failure for Sparkfun Thing Dev (ESP8266) when using WiFiUDP

ThomasBleijendaal, 2016-10-12:
### Configuration

**Operating system**:
Windows 10 Home

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.10.3

### Description of problem
Cannot build anything for an ESP8266.

I was trying to test a new program for the Sparkfun Thing Dev, but could not build anything anymore after I updated PlatformIO and the espressif platform to the newest version. I removed everything from my program to isolate the problem, and traced the problem down to the WiFiUdp of the espressif platform. If you comment out the ```WiFiUDP Udp``` the build succeeds. 

#### Steps to Reproduce

1. Make new project, IDE: Visual Studio, Platform: Sparkfun Thing Dev.
2. Create main.cpp and copy code below.
3. Run platformio run.

### Actual Results
Build log:
```
[10/12/16 22:46:17] Processing thingdev (platform: espressif, upload_port: COM3, board: thingdev, framework: arduino)
-----------------------------------------------------------------------------------------------------------------------
xtensa-lx106-elf-g++ -o .pioenvs\thingdev\firmware.elf -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -Wl,-T"esp8266.flash.512k64.ld" .pioenvs\thingdev\src\main.o -LC:\Users\thoma\.platformio\packages\framework-arduinoespressif\tools\sdk\lib -LC:\Users\thoma\.platformio\packages\ldscripts -L.pioenvs\thingdev -Wl,--start-group .pioenvs\thingdev\libFrameworkArduinoVariant.a .pioenvs\thingdev\libFrameworkArduino.a -lmesh -lwpa2 -lsmartconfig -lpp -lmain -lwpa -llwip -lnet80211 -lwps -lcrypto -lphy -lhal -laxtls -lgcc -lm .pioenvs\thingdev\libESP8266WiFi.a -Wl,--end-group
.pioenvs\thingdev\libESP8266WiFi.a(ESP8266WiFiGeneric.o):(.text._ZN23ESP8266WiFiGenericClass14_eventCallbackEPv+0x4): undefined reference to `std::__detail::_List_node_base::_M_unhook()'
.pioenvs\thingdev\libESP8266WiFi.a(ESP8266WiFiGeneric.o): In function `ESP8266WiFiGenericClass::_eventCallback(void*)':
ESP8266WiFiGeneric.cpp:(.text._ZN23ESP8266WiFiGenericClass14_eventCallbackEPv+0x64): undefined reference to `std::__detail::_List_node_base::_M_unhook()'
collect2.exe: error: ld returned 1 exit status
scons: *** [.pioenvs\thingdev\firmware.elf] Error 1
============================================= [ ERROR ] Took 1.99 seconds =============================================
```

### Expected Results
Successful build.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:thingdev]
platform = espressif
framework = arduino
board = thingdev
upload_port = COM3
```

**Source file to reproduce issue:**
```main.cpp
#include "Arduino.h"
#include "ESP8266WiFi.h"
#include "WiFiUdp.h"

WiFiUDP Udp;

void setup() {
	Serial.begin(9600);
}
void loop() {
	Serial.println("Jeej");
	delay(1000);
}
```

### Additional info

ivankravets, 2016-10-12:
Does it work in Arduino IDE?

ThomasBleijendaal, 2016-10-12:
Yes it does. Board version in Arduino ID 2.3.0, in Platform IO:

```
pio platforms show espressif
espressif   - Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications. [ https://espressif.com/ ]
----------
Package: toolchain-xtensa
Alias: toolchain
Version: 2
Installed: 2016-06-22 17:40:13
----------
Package: tool-esptool
Alias: uploader
Version: 7
Installed: 2016-06-22 17:40:13
----------
Package: tool-mkspiffs
Alias: uploader
Version: 1
Installed: 2016-06-22 21:45:38
----------
Package: framework-arduinoespressif
Alias: framework
Version: 14
Installed: 2016-10-10 23:13:11
----------
Package: ldscripts
Version: 23
Installed: 2016-10-10 23:13:11
```

ivankravets, 2016-10-12:
We don't support PlatformIO 2.0

Please `pio upgrade` to PlatformIO 3.0

closed 2016-10-12 by ivankravets