platformio/platformio-core #803: ValueError: No JSON object could be decoded

seanlerner, 2016-10-13:
### Configuration

**Operating system**: raspberry pi / Raspbian GNU/Linux 8.0 (jessie)

**PlatformIO Version** (`platformio --version`): 3.1.0 and 3.0.1

### Description of problem

I receive the error "ValueError: No JSON object could be decoded".

Previously, platformio was working on my raspberry pi, but I recently tried the develop branch out, it didn't work, so I switched back to the stable branch. And now the stable branch always produces this error. I tried v3.0.1 as well and it failed. 

I've just switched to v2.11.2 and it's working again.

#### Steps to Reproduce

sudo platformio run --target upload

### Actual Results

```
[Wed Oct 12 23:20:51 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 106, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/run.py", line 108, in cli
    results[envname] = ep.process()
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/run.py", line 191, in process
    result = self._run()
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/run.py", line 283, in _run
    return p.run(build_vars, build_targets, self.silent, self.verbose)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/platform.py", line 244, in run
    self.install_packages(silent=True)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/platform.py", line 203, in install_packages
    self.pm.install(name, opts.get("version"), silent=silent)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/package.py", line 425, in install
    package_dir = self.get_package_dir(name, requirements, url)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/package.py", line 402, in get_package_dir
    package = self.get_package(name, requirements, url)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/package.py", line 372, in get_package
    for manifest in self.get_installed():
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/package.py", line 354, in get_installed
    manifest = self.load_manifest(pkg_dir)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/package.py", line 143, in load_manifest
    manifest = util.load_json(path)
  File "/usr/local/lib/python2.7/dist-packages/platformio/util.py", line 158, in load_json
    return json.load(f)
  File "/usr/lib/python2.7/json/__init__.py", line 290, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

[... 265 lines cut ...]
        break;
      case semicolon:
        execCommand();
        break;
      case backSpace:
        if (command.length() > 0) {
          command.remove(command.length() - 1);
          Serial.print("\b \b");
        }
        break;
      default:
        incomingChar = char(incomingByte);
        Serial.print(incomingChar);
        command.concat(incomingChar);
        break;
    }
  }
}
```

### Additional info

Things are working again with 2.11.2 so I'm happy but thought I should report this. Thanks for all your hard work.

Sean

ivankravets, 2016-10-13:
Try to remove totally `~/.platformio` folder and start again. The problem with broken manifest for package.

seanlerner, 2016-10-13:
@ivankravets I removed ~/.platformio but I noticed I also had a /root/.platformio so I removed that as well and it resolved the issue. Thanks a lot.

closed 2016-10-13 by seanlerner