platformio/platformio-core #804: Could you support Pololu's Astar32U4

crazyquark, 2016-10-13:
These are the boards:
[AStar32U4](https://www.pololu.com/product/3101)

And the code:
[AStar32U4](https://www.pololu.com/product/3101)

ivankravets, 2016-10-13:
The library is added http://platformio.org/lib/show/622/AStar32U4

Please use temporary Arduino Leonardo (`leonardo`) board configuratrion. They are the same.

crazyquark, 2016-10-17:
I did do that however I can't figure out how to set the board's USB to act as a keyboard?
In Arduino IDE this was by default.

crazyquark, 2016-10-18:
Using build_flags = -DUSB_SERIAL_HID or -DUSB_HID seems to have no effect on this board.

ivankravets, 2016-10-20:
The repo is created. Please make PR to it.

https://github.com/platformio/platformio-pkg-framework-arduinoavr

ivankravets, 2016-11-18:
Fixed in https://github.com/platformio/platform-atmelavr/commit/5af0f34c89a1a23b5b36f9411a18d61665c24da4

closed 2016-11-18 by ivankravets