platformio/platformio-core #805: Cannot upload to mzeropro with using stable or development version

fjammes, 2016-10-14:
### Configuration

**Operating system**:

Debian jessie

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 3.2.0a7

### Description of problem

Can not upload to mzeropro because of minor openocd error:

```shell
Uploading .pioenvs/mzeropro/firmware.bin
Open On-Chip Debugger 0.9.0-gd4b7679 (2014-10-03-00:26)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
/home/lora/.platformio/packages/tool-openocd/bin/openocd.bin: option requires an argument -- 'c'
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.1F.0118
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
```

#### Steps to Reproduce

1. git clone git@github.com:fjammes/braise.git
2. cd braise/gps_bme_send
3. platformio init --board mzeropro
4. platformio run -t upload

### Actual Results

Script waits without doing nothing.

### Expected Results

Upload code to mzeropro

### Additional info

Changing openocd '-c' with '--command=' option works fine, as shown below:
```
~/.platformio/packages/tool-openocd/bin/openocd  -f /home/lora/.platformio/packages/framework-arduinosam/variants/arduino_zero_org/openocd_scripts/arduino_zero_org.cfg -s /home/lora/.platformio/packages/tool-openocd/share/openocd/scripts -s /home/lora/.platformio/packages/tool-openocd/share/openocd/scripts/board -s /home/lora/.platformio/packages/framework-arduinosam/variants/arduino_zero_org/openocd_scripts --command="telnet_port disabled; program {{.pioenvs/mzeropro/firmware.bin}} verify reset 0x4000; shutdown"
# Command stdout:
Open On-Chip Debugger 0.9.0-gd4b7679 (2014-10-03-00:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.1F.0118
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000028f4 msp: 0x20002c00
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
wrote 49152 bytes from file .pioenvs/mzeropro/firmware.bin in 6.412093s (7.486 KiB/s)
** Programming Finished **
** Verify Started **
verified 48948 bytes in 3.957394s (12.079 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

ivankravets, 2016-10-14:
@valeros I have the same issue. Could you help us?

@fjammes Please use `mzeroproUSB` board and USB port instead. It works.

fjammes, 2016-10-17:
Thanks, I can indeed use mzeroproUSB to upload binaries, but then `platformio device monitor -p /dev/ttyACM0` does not print any output message.

fjammes, 2016-10-17:
I would need to use SerialUSB.print(...) in my code, but some third-parties use Serial.print(), so it seems I'm stick with programming port. Coudl you please fix this bug?

ivankravets, 2016-10-17:
Could you try one fix? Please go to `~/.platformio/packages/tool-openocd/bin/` and edit `openocd` wrapper. Replace 
```
LD_LIBRARY_PATH="$MYAPP_HOME/lib" "$0.bin" "$1" "$2" "$3" "$4" "$5" "$6" "$7" "$8" "$9"
```
with
```
LD_LIBRARY_PATH="$MYAPP_HOME/lib" "$0.bin" "$@"
```

Does it work now?

ivankravets, 2016-10-26:
@fjammes do you have any news?

ivankravets, 2016-11-18:
Just run `pio update` where `tool-openocd` should be updated.

/cc @fraimondo @fjammes

closed 2016-11-18 by ivankravets

fraimondo, 2016-11-18:
What was problem/solution then?

ivankravets, 2016-11-18:
@fraimondo  see https://github.com/platformio/platformio/issues/805#issuecomment-254199812

commit 3fa8b45e (2016-11-18): Fix issue with uploading to Arduino Zero on programming USB port // Resolve #805 Resolve #828