platformio/platformio-core #806: PLATFORMIO_HOME_DIR not used well for eclipse init

milossoucek, 2016-10-14:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform. All issues related to Development Platform should be reported to appropriate repository. Search it using link below
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: windows7

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.1.0

### Description of problem
When default platformio home directory is used ( under user profile ), initialization of project for eclipse works well.
But when I set up environment variable PLATFORMIO_HOME_DIR, then initialization of project for eclipse is not correct - "compiler" headers (like stdlib.h) are not visible in eclipse for code completion etc.

The source of problem is in generated file language.settings.xml, where path to compiler is wrong - it still includes ${USERPROFILE}, but it should not.

e.g. when I set up: PLATFORMIO_HOME_DIR=C:\PIO\IDE\HOME_DIR
then in project file language.settings.xml is line containing:
....name="CDT Cross GCC Built-in Compiler Settings" parameter="${USERPROFILE}C:\PIO\IDE\HOME_DIR\packages\toolchain....

tested with project setup for avr/uno and esp8266/d1-mini with same result

ivankravets, 2018-02-13:
Could you reproduce this issue with the latest http://docs.platformio.org/en/latest/installation.html#development-version ?

milossoucek, 2018-02-14:
Hi,
Uninstalled previous version with my modifications, and installed new development one. Tested on arduino uno and stm32 projects. Uno looks working fine, but STM still have the same problem.
Here is content of file from stm32 project:
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<project>
	<configuration id="0.910961921" name="Default">
		<extension point="org.eclipse.cdt.core.LanguageSettingsProvider">
			<provider copy-of="extension" id="org.eclipse.cdt.ui.UserLanguageSettingsProvider"/>
			<provider-reference id="org.eclipse.cdt.core.ReferencedProjectsLanguageSettingsProvider" ref="shared-provider"/>
			<provider-reference id="org.eclipse.cdt.managedbuilder.core.MBSLanguageSettingsProvider" ref="shared-provider"/>
			<provider class="org.eclipse.cdt.internal.build.crossgcc.CrossGCCBuiltinSpecsDetector" console="false" env-hash="1291887707783033084" id="org.eclipse.cdt.build.crossgcc.CrossGCCBuiltinSpecsDetector" keep-relative-paths="false" name="CDT Cross GCC Built-in Compiler Settings" parameter="${USERPROFILE}C:\PIO\IDE\pio_home\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe ${FLAGS} -E -P -v -dD &quot;${INPUTS}&quot;" prefer-non-shared="true">
				<language-scope id="org.eclipse.cdt.core.gcc"/>
				<language-scope id="org.eclipse.cdt.core.g++"/>
			</provider>
		</extension>
	</configuration>
	<configuration id="0.910961921.1363900502" name="Debug">
		<extension point="org.eclipse.cdt.core.LanguageSettingsProvider">
			<provider copy-of="extension" id="org.eclipse.cdt.ui.UserLanguageSettingsProvider"/>
			<provider-reference id="org.eclipse.cdt.core.ReferencedProjectsLanguageSettingsProvider" ref="shared-provider"/>
			<provider-reference id="org.eclipse.cdt.managedbuilder.core.MBSLanguageSettingsProvider" ref="shared-provider"/>
			<provider class="org.eclipse.cdt.internal.build.crossgcc.CrossGCCBuiltinSpecsDetector" console="false" env-hash="1291887707783033084" id="org.eclipse.cdt.build.crossgcc.CrossGCCBuiltinSpecsDetector" keep-relative-paths="false" name="CDT Cross GCC Built-in Compiler Settings" parameter="${USERPROFILE}C:\PIO\IDE\pio_home\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe ${FLAGS} -E -P -v -dD &quot;${INPUTS}&quot;" prefer-non-shared="true">
				<language-scope id="org.eclipse.cdt.core.gcc"/>
				<language-scope id="org.eclipse.cdt.core.g++"/>
			</provider>
		</extension>
	</configuration>
</project>

And here is content of file for UNO project:
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<project>
	<configuration id="0.910961921" name="Default">
		<extension point="org.eclipse.cdt.core.LanguageSettingsProvider">
			<provider copy-of="extension" id="org.eclipse.cdt.ui.UserLanguageSettingsProvider"/>
			<provider-reference id="org.eclipse.cdt.core.ReferencedProjectsLanguageSettingsProvider" ref="shared-provider"/>
			<provider-reference id="org.eclipse.cdt.managedbuilder.core.MBSLanguageSettingsProvider" ref="shared-provider"/>
			<provider class="org.eclipse.cdt.internal.build.crossgcc.CrossGCCBuiltinSpecsDetector" console="false" env-hash="-592958367594185759" id="org.eclipse.cdt.build.crossgcc.CrossGCCBuiltinSpecsDetector" keep-relative-paths="false" name="CDT Cross GCC Built-in Compiler Settings" parameter="C:\PIO\IDE\pio_home\packages\toolchain-atmelavr\bin\avr-g++.exe ${FLAGS} -E -P -v -dD &quot;${INPUTS}&quot;" prefer-non-shared="true">
				<language-scope id="org.eclipse.cdt.core.gcc"/>
				<language-scope id="org.eclipse.cdt.core.g++"/>
			</provider>
			<provider copy-of="extension" id="org.eclipse.cdt.managedbuilder.core.GCCBuildCommandParser"/>
		</extension>
	</configuration>
	<configuration id="0.910961921.1363900502" name="Debug">
		<extension point="org.eclipse.cdt.core.LanguageSettingsProvider">
			<provider copy-of="extension" id="org.eclipse.cdt.ui.UserLanguageSettingsProvider"/>
			<provider-reference id="org.eclipse.cdt.core.ReferencedProjectsLanguageSettingsProvider" ref="shared-provider"/>
			<provider-reference id="org.eclipse.cdt.managedbuilder.core.MBSLanguageSettingsProvider" ref="shared-provider"/>
		</extension>
	</configuration>
</project>

Even when i completely delete the files, then after run of "Rebuild C/C++ project Index" in eclipse the content is generated the same.

milossoucek, 2018-02-14:
For any reason, my paste of xml content is ignored.
So here they are as files.

[language.settings.xml stm32.txt](https://github.com/platformio/platformio-core/files/1724951/language.settings.xml.stm32.txt)
[language.settings.xml.uno.txt](https://github.com/platformio/platformio-core/files/1724952/language.settings.xml.uno.txt)

commit 412a1f78 (2019-07-04): Fixed an issue when generating invalid "Eclipse CDT Cross GCC Built-in Compiler Settings" if a custom PLATFORMIO_CORE_DIR is used // Resolve #806
closed 2019-07-04 by ivankravets