platformio/platformio-core #807: cant install Platformio

jourjine, 2016-10-15:
Windows 7 x64 Russian 
all regional setting - USA
always have error:

Traceback (most recent call last):
  File "C:\Users\Dmitry\AppData\Local\Temp\d-116915-1352-jfy0tp.1kp3oc4n29\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\Dmitry\AppData\Local\Temp\d-116915-1352-jfy0tp.1kp3oc4n29\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\Dmitry\AppData\Local\Temp\d-116915-1352-jfy0tp.1kp3oc4n29\virtualenv-14.0.6\virtualenv.py", line 900, in create_environment
    home_dir, lib_dir, inc_dir, bin_dir = path_locations(home_dir)
  File "C:\Users\Dmitry\AppData\Local\Temp\d-116915-1352-jfy0tp.1kp3oc4n29\virtualenv-14.0.6\virtualenv.py", line 944, in path_locations
    mkdir(home_dir)
  File "C:\Users\Dmitry\AppData\Local\Temp\d-116915-1352-jfy0tp.1kp3oc4n29\virtualenv-14.0.6\virtualenv.py", line 317, in mkdir
    os.makedirs(path)
  File "C:\Python27\lib\os.py", line 157, in makedirs
    mkdir(name, mode)
WindowsError: [Error 5] ???????? ? ???????: 'C:\\Users\\Dmitry\\.atom\\packages\\platformio-ide\\penv'

ivankravets, 2016-10-26:
"Error 5: Access is denied". Could you delete this folder `C:\\Users\\Dmitry\\.atom\\packages\\platformio-ide\` and install PlatformIo IDE?

ivankravets, 2016-11-14:
Please reopen issue if you still need a help

closed 2016-11-14 by ivankravets