platformio/platformio-core #808: Vim: linter can't find the 'Arduino.h' file

lynndylanhurley, 2016-10-16:
What kind of issue is this?

- [x] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**:

Mac OS X 10.11.6

**PlatformIO Version** (`platformio --version`):

`3.1.0`

### Description of problem

Using Vim, the [neomake](https://github.com/neomake/neomake) linter plugin can't seem to find the Arduino libs. [Here is the complete output of the error](https://gist.github.com/lynndylanhurley/de7fdab016da1c72842d39e54a6e49c7).

This is the command that is run:
```
clang++ -fsyntax-only -Wall -Wextra /Users/lynnhurley/Code/ChainedSignal
s/surface_event_display/src/main.cpp
```

And this is the error:
```
clangtidy: 'Arduino.h' file not found [clang-diagnostic-error]
```

`platformio run` works just fine.

#### Steps to Reproduce

1. `platformio init --board megaatmega2560`
2. Paste blink code from the [quickstart guide](http://docs.platformio.org/en/stable/quickstart.html#initialize-project).
3. Save `src/main.cpp`

### Actual Results

The following error:
```
clangtidy: 'Arduino.h' file not found [clang-diagnostic-error]
```

### Expected Results

There should be no linter errors.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
```

**Source file to reproduce issue:**
```cpp
/**
 * Blink
 *
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */
#include "Arduino.h"

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);

  // wait for a second
  delay(1000);

  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);

  // wait for a second
  delay(1000);
}
```

### Additional info

I guess the main questions are:

1. The `neomake` linter wants to use `clang++` by default. Is this correct, or should `neomake` be using a different system for linting?
2. If `clang++` is the correct linter, what is the best way to tell `clang++` where the `platformio` libraries are located?
3. If `clang++` is not the correct linter, what commands should `neomake` be running instead?

ivankravets, 2016-10-26:
You see this error because didn't specify CPPPATH. 

PlatformIO Build System has this data and other IDEs use it. For example, check `pio run -t idedata`. Does `neomake` allow to specify custom CPPPATH?

lynndylanhurley, 2016-10-26:
Oh nice, I didn't know about that command. I'll see what I can do with that, thanks @ivankravets!!

coddingtonbear, 2016-12-04:
I'm sure you've already found a suitable answer, @lynndylanhurley, but for future readers, I've written a simple plugin for solving this problem in a general way.

* [neomake-platformio](https://github.com/coddingtonbear/neomake-platformio)

Cheers!

ivankravets, 2016-12-04:
@coddingtonbear thanks a lot for your plugin! I've just added it to PIO docs http://docs.platformio.org/en/latest/ide/vim.html

commit 4efc0c04 (2016-12-04): Native integration with VIM/Neovim using “neomake-platformio” plugin // Resolve #467 Resolve #808
closed 2016-12-04 by ivankravets