platformio/platformio-core #810: Eclipse project has its own system path defined

milossoucek, 2016-10-17:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform. All issues related to Development Platform should be reported to appropriate repository. Search it using link below
      https://github.com/platformio?query=platform-

- [x] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: windows 7, but looks OS independend

**PlatformIO Version** (`platformio --version`): 3.1.0

### Description of problem
When initialising a project for Eclipse, also PATH is overriden (defined specific for eclipse project) in project settings. I think this is unhappy, since it prevents usage of the same eclipse project in different environments (e.g. on different PCs with different PIO instalation directories). This should be avoided and only system path should be used instead. It also prevents further "init" of project in Eclipse after environment change, you have to manually delete the path first in project settings, then it can be re-initialised ( and gets a new path defined in project again :( ).

The path override is defined in generated file .settings/org.eclipse.cdt.core.prefs
And it can be well visible in "source template" platformio\ide\tpls\eclipse\.settings\org.eclipse.cdt.core.prefs.tpl 

If user is able to run PIO manually, then his system path is already set up. So it looks not necessary to do it again in Eclipse project. It can maybe be seen as a bug ...

milossoucek, 2016-10-17:
Additional hint: the file .settings/org.eclipse.cdt.core.prefs can be NOT generated at all. This will 1) not overwrite the path and 2) it will allow user to define his own environment for project in Eclipse, which will not be overwritten by any further PIO init. As tested with Eclipse Neon.1, the file is not needed for project to work, and is written by Eclipse only if user changes any setting stored in this file.

ivankravets, 2016-10-17:
Eclipse starts with invalid(system) $PATH by default. In this case, PlatformIO doesn't work at all. That is why we set own PATH with prepended PATH to PIO binary.

We have similar issue https://github.com/platformio/platformio/issues/443

milossoucek, 2016-10-17:
I must say I do not understand. When I run Eclipse, it has set up a path, which is equal to actual system path. And this is where also PIO must be working, otherwise user will not be able to run PIO as CLI.
If user has set up environment for PIO, then running eclipse under it must work - all settings (envs) are take over as default.

I did tests on it, and it looks like there is no reason to have PATH set in project, mainly if it is the same as path already set in system ...

Issue 443 looks not very simillar, only that it is also about Eclipse setup :) Or you mean that missing file in template will cause deleted file in Eclipse setup during init ? Then, template can be simplified removing the path setup from it, and not by deleting the file completely.

ivankravets, 2016-10-17:
I mean Unix OS. User install PIO in local environment, not to system.

443 means that need to update only required fields but no need to regenerate whole project and override custom settings.

milossoucek, 2016-10-17:
Thanks, now it is more clear.
Maybe than it makes sense to generate this file only in case it is not already present and/or write path only if the path is different from actually needed one ( = with functional PIO )

milossoucek, 2016-10-17:
Btw. - by UNIX you mean any linux too, right ? Or you mean only UNIX systems ?

milossoucek, 2016-10-18:
OK, after looking on web I have seen many pople have problem with path on non-windows OSes.
This brings possible solution for pio:
How about to add this path to project only in case project is not on windows ?

ivankravets, 2016-10-18:
Could you explain me why do you need different versions of PlatformIO? You say that you have multiple PIO in different directories. Thanks.

milossoucek, 2016-10-18:
Imagine full setup PIO project in eclipse, working well on one PC.
Then, when you move it ( e.g. give to a friend as start point for his play ), he can not start working from eclipse when he has different PATH, e.g. installed python somewhere else, or have different folder for windows instalation or whatever else.
And this looks like a problem, I think.
Without setting the PATH in project, you will not face such issue.

And my true full story: :)
I do not have multiple PIO instalations, but I have one on flash disk moving with me. :)
Atom looks like a problem to be portable, and because I am using eclipse in work, I start playing with eclipse as IDE. Its portability is very good, and together with winPython it is ready for PIO.
I run eclipse from batch file, which sets up environment according to current folder and drive.
And then there are two problems which I found.
One is this, and another is issue #806. That issue I solved by setting my own %USERPROFILE%, which is horrible hack, but looks working for eclipse+pio.

Solving both, we can have nice portable PIO+Eclipse environment set up and ready to use just by unzipping :)

ivankravets, 2016-10-18:
> I do not have multiple PIO instalations, but I have one on flash disk moving with me. :)

:eyes: PlatformIO on flash disk? :eyes: :smiley_cat:

The right solution is #443 that means that need to update only modified fields of project config. 

P.S: You can remove [this file](https://github.com/platformio/platformio/blob/develop/platformio/ide/tpls/eclipse/.settings/org.eclipse.cdt.core.prefs.tpl) from PIO sources after installation.

milossoucek, 2016-10-19:
Thanks a lot, good hint.
I did not catch till now that all the sources are still available after install, I thought it is installing just the exe which contain everything. This changes a lot, I can play much more with it :)

Btw. yesterday I was playing again with atom, and I found that actual revision is portable !
So I did other folder on flash disk with atom, and it looks working well.
It has big advantage that it is really well working together with PIO, maybe I will switch to it from eclipse :) 
Good job, Ivan !

closed 2019-07-04 by ivankravets

ivankravets, 2019-07-04:
We did some change. Now, we append the original $PATH to an overridden variable. We need this variable because this is a problem of Eclipse on Unix systems. It does not handle environment from changed shell.

What is more, I don't see any problem. Current integration supposes that end developers will re-run `pio init --ide eclipse` in the project with the each change when add new library or change project structure. So, in this case, PIO will update valid PATH

commit c69e8024 (2019-07-04): Append system PATH to overridden in CDT project // Resolve #810