platformio/platformio-core #811: For Leonardo board: how to enable USB HID behavior?

crazyquark, 2016-10-18:
I tried using -DUSB_HID and -DUSB_SERIAL_HID in the build_flags section but it hasn't worked:
[env:leonardo]
platform = atmelavr
board = leonardo
framework = arduino
build_flags = -DUSB_SERIAL_HID

My board is an Astar32u4(from Pololu)

ivankravets, 2016-10-18:
Please go to Arduino IDE settings and enable verbose output for building and uploading.
Then build project in Arduino IDE and share an output on http://pastebin.com

crazyquark, 2016-10-18:
Here is the compile output:
http://pastebin.com/pdMqWy6B

And the upload output:
http://pastebin.com/PLRuXeDt

ivankravets, 2016-10-18:
Try to add these macros
```ini
[env:leonardo]
platform = atmelavr
board = leonardo
framework = arduino
build_flags = -DARDUINO_AVR_A_STAR_32U4 -DUSB_VID=0x1ffb -DUSB_PID=0x2300 -DUSB_MANUFACTURER="Pololu Corporation" -DUSB_PRODUCT="Pololu A-Star 32U4"
```

Check PIO build output that flags are set properly via `pio run --verbose`.

crazyquark, 2016-10-20:
Unfortunately that had no effect.
The device is recognized in the Device Manager as "Arduino leonardo bootloader", so only serial communication is possible.

ivankravets, 2016-11-18:
Fixed in https://github.com/platformio/platform-atmelavr/commit/5af0f34c89a1a23b5b36f9411a18d61665c24da4

closed 2016-11-18 by ivankravets