platformio/platformio-core #812: extra_script preand post action doesn't work with buildprog

dehav, 2016-10-22:
Tried to use pre/post action for 'buildprog' in an extra_script per example in doc. While it worked for 'size', it didn't work for 'buildprog'. FYI, the line

print "Current build targets", map(str, BUILD_TARGETS)

printed 

Current build targets ["buildprog", "size"]

ivankravets, 2016-10-27:
All development platforms have been updated too. See
https://github.com/ivankravets?tab=overview&from=2016-10-27

For example, https://github.com/platformio/platform-atmelavr/commit/5b438b9b5428dcfce39b60d558bfc2447db4cc82

New development platforms will be released with PlatformIO 3.2.

Disane87, 2017-12-25:
Issue persists. Any solutions for that?

Disane87, 2017-12-29:
I just discovered, that this isn‘t an issue of your framework. It was a failure by me. I didn‘t changed anything in the main.cpp so the compiler doesn‘t trigger the buildprog. The size is called without an problem. 

I guess it would be great, to implement a hint like: „nothing changed, buildprog is not called“

ivankravets, 2017-12-29:
That is normal behavior. If you need to call something "always", please use [AlwaysBuild](http://scons.org/doc/production/HTML/scons-user.html#idm139933296955680).

closed 2017-12-29 by ivankravets

Disane87, 2017-12-29:
Yep, sorry for bothing you! :)

Piroro-hs, 2018-03-03:
Issue persists for me. I checked that I changed the content in the main.cpp...

ivankravets, 2018-03-03:
@Piroro-hs could you share your `platformio.ini` and extra script to reproduce this issue?

dehav, 2018-03-03:
If your asking me, no I can't. I've been removed from this for too long now. Sorry.
  ----- Original Message ----- 
  From: Ivan Kravets 
  To: platformio/platformio-core 
  Cc: dehav ; Author 
  Sent: Saturday, March 03, 2018 3:15 AM
  Subject: Re: [platformio/platformio-core] extra_script preand post action doesn't work with buildprog (#812)

  @Piroro-hs could you share your platformio.ini and extra script to reproduce this issue?

  —

       Virus-free. www.avg.com

Piroro-hs, 2018-03-03:
Sorry, it works. I've just misunderstood that the pre callback are excuted before compiling starts.

commit 53c4c6d2 (2016-10-27): Update docs with "buildprog" target, add/post actions // Resolve #812