platformio/platformio-core #813: Errno 111 when initializing on a Debian PC with strict Proxy Settings.

RebelCoderRU, 2016-10-24:
Hi.

I have installed PIO on my Debian 8.6 Linux, and it fails to initialize. This VM is under very strict Proxy settings, but is set up to work with this proxy as all other apps and things work just right.

```
jurisl@YuriyLinuxVM:~/Work$ python get-platformio.py

==> Installing Python Package Manager ...
<urlopen error [Errno 111] Connection refused>
[FAILURE]
The installation process has been FAILED!
Please report about this problem here
< https://github.com/platformio/platformio/issues >
jurisl@YuriyLinuxVM:~/Work$ 
```

[image]

ivankravets, 2016-10-24:
Hi,

Could you try to install using `pip` package manager?

```
cd /tmp
virtualenv piotest
source piotest/bin/activate
pip install platformio
```

RebelCoderRU, 2016-10-24:
Иван, тоже не сработало.

I have tried your suggestion and it fails too.

```
jurisl@YuriyLinuxVM:/tmp$ virtualenv piotest
Running virtualenv with interpreter /usr/bin/python2
New python executable in piotest/bin/python2
Also creating executable in piotest/bin/python
Installing setuptools, pip...done.
jurisl@YuriyLinuxVM:/tmp$ source piotest/bin/activate
(piotest)jurisl@YuriyLinuxVM:/tmp$ pip install platformio
Downloading/unpacking platformio
  Cannot fetch index base URL https://pypi.python.org/simple/
  Could not find any downloads that satisfy the requirement platformio                                       
Cleaning up...
No distributions at all found for platformio
Storing debug log for failure in /home/jurisl/.pip/pip.log
```
[pip.log.txt](https://github.com/platformio/platformio/files/548212/pip.log.txt)

P.S. On exactly the same system at home (with no proxy), everything works perfectly. On this PC I have a corporate proxy and it fails to install. Let me know what you think.

Thanks.

ivankravets, 2016-10-24:
How about https://www.google.com.ua/search?q=pip+proxy?

RebelCoderRU, 2016-10-24:
Just got it working with proxy, and init works:

j```
urisl@YuriyLinuxVM:~$ sudo platformio init

The current working directory /home/jurisl will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories have been created in /home/jurisl
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries

Project has been successfully initialized!
Useful commands:
`platformio run` - process/build project from the current directory
`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board
`platformio run --target clean` - clean project (remove compiled files)
`platformio run --help` - additional information

```

Atom IDE still fails with the same error as above (picture). I will use Qt Creator for now though.

ivankravets, 2016-10-24:
Cool! Now, let's fix Atom. What did you do to install PlatformIO above? Looks like need to setup PROXY environment variables and restart PC.

RebelCoderRU, 2016-10-24:
This Linux has been configured to use Proxy, and everything else works just fine with it. For some reason Atom IDE does not pick it up and fails. I used .deb package to install it. I will keep hacking now to fix it.

RebelCoderRU, 2016-10-24:
I got it a bit further, but it still fails on a last step. It updates Atom, can search for packages, so it has the connection working.

[image]

[image]

[image]

ivankravets, 2016-10-25:
Clone [this](https://github.com/platformio/platformio-atom-ide) repo to `~/.atom/packages/platformio-ide`, change directory to it and run `npm install`. Open Atom.

RebelCoderRU, 2016-10-25:
Followed your instructions, and error is not showing up in Atom anymore, platformio-ide seems to be installed but it did not integrate into Atom at all.

```
jurisl@YuriyLinuxVM:~/.atom/packages/platformio-ide/platformio-atom-ide$ sudo npm install
npm WARN engine dexie@1.5.0: wanted: {"node":">=4.2"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine form-data@2.1.1: wanted: {"node":">= 0.12"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine hawk@3.1.3: wanted: {"node":">=0.10.32"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine cryptiles@2.0.5: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine boom@2.10.1: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine hoek@2.16.3: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine string-width@2.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine is-fullwidth-code-point@2.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine co@4.6.0: wanted: {"iojs":">= 1.0.0","node":">= 0.12.0"} (current: {"node":"0.10.29","npm":"1.4.21"})
ini@1.3.4 node_modules/ini

semver@5.3.0 node_modules/semver

lockfile@1.0.2 node_modules/lockfile

command-join@1.1.1 node_modules/command-join
├── repeat-string@1.6.1
└── array-from@2.1.1

temp@0.8.3 node_modules/temp
├── os-tmpdir@1.0.2
└── rimraf@2.2.8

promisify-node@0.4.0 node_modules/promisify-node
├── object-assign@4.1.0
└── nodegit-promise@4.0.0 (asap@2.0.5)

fs-extra@0.26.7 node_modules/fs-extra
├── path-is-absolute@1.0.1
├── klaw@1.3.0
├── jsonfile@2.4.0
├── graceful-fs@4.1.9
└── rimraf@2.5.4 (glob@7.1.1)

tar@2.2.1 node_modules/tar
├── inherits@2.0.3
├── block-stream@0.0.9
└── fstream@1.0.10 (graceful-fs@4.1.9, mkdirp@0.5.1, rimraf@2.5.4)

request@2.76.0 node_modules/request
├── is-typedarray@1.0.0
├── oauth-sign@0.8.2
├── aws-sign2@0.6.0
├── forever-agent@0.6.1
├── tunnel-agent@0.4.3
├── caseless@0.11.0
├── stringstream@0.0.5
├── isstream@0.1.2
├── json-stringify-safe@5.0.1
├── extend@3.0.0
├── aws4@1.5.0
├── node-uuid@1.4.7
├── combined-stream@1.0.5 (delayed-stream@1.0.0)
├── qs@6.3.0
├── form-data@2.1.1 (asynckit@0.4.0)
├── tough-cookie@2.3.1
├── mime-types@2.1.12 (mime-db@1.24.0)
├── har-validator@2.0.6 (pinkie-promise@2.0.1, commander@2.9.0, chalk@1.1.3, is-my-json-valid@2.15.0)
├── http-signature@1.1.1 (assert-plus@0.2.0, jsprim@1.3.1, sshpk@1.10.1)
└── hawk@3.1.3 (cryptiles@2.0.5, sntp@1.0.9, boom@2.10.1, hoek@2.16.3)

dexie@1.5.0 node_modules/dexie

babel-eslint@6.1.2 node_modules/babel-eslint
├── lodash.assign@4.2.0
├── lodash.pickby@4.6.0
├── babylon@6.13.0
├── babel-types@6.18.0 (to-fast-properties@1.0.2, esutils@2.0.2, lodash@4.16.4, babel-runtime@6.18.0)
└── babel-traverse@6.18.0 (babel-messages@6.8.0, globals@9.12.0, debug@2.2.0, invariant@2.2.1, babel-code-frame@6.16.0, lodash@4.16.4, babel-runtime@6.18.0)

eslint@2.13.1 node_modules/eslint
├── path-is-inside@1.0.2
├── path-is-absolute@1.0.1
├── imurmurhash@0.1.4
├── strip-json-comments@1.0.4
├── pluralize@1.2.1
├── ignore@3.2.0
├── progress@1.1.8
├── globals@9.12.0
├── estraverse@4.2.0
├── text-table@0.2.0
├── esutils@2.0.2
├── user-home@2.0.0 (os-homedir@1.0.2)
├── is-resolvable@1.0.0 (tryit@1.0.2)
├── debug@2.2.0 (ms@0.7.1)
├── chalk@1.1.3 (escape-string-regexp@1.0.5, supports-color@2.0.0, ansi-styles@2.2.1, has-ansi@2.0.0, strip-ansi@3.0.1)
├── require-uncached@1.0.2 (resolve-from@1.0.1, caller-path@0.1.0)
├── json-stable-stringify@1.0.1 (jsonify@0.0.0)
├── levn@0.3.0 (type-check@0.3.2, prelude-ls@1.1.2)
├── mkdirp@0.5.1 (minimist@0.0.8)
├── doctrine@1.5.0 (isarray@1.0.0)
├── shelljs@0.6.1
├── glob@7.1.1 (fs.realpath@1.0.0, inherits@2.0.3, inflight@1.0.6, once@1.4.0, minimatch@3.0.3)
├── optionator@0.8.2 (fast-levenshtein@2.0.5, deep-is@0.1.3, wordwrap@1.0.0, type-check@0.3.2, prelude-ls@1.1.2)
├── concat-stream@1.5.2 (inherits@2.0.3, typedarray@0.0.6, readable-stream@2.0.6)
├── is-my-json-valid@2.15.0 (generate-function@2.0.0, jsonpointer@4.0.0, xtend@4.0.1, generate-object-property@1.2.0)
├── file-entry-cache@1.3.1 (object-assign@4.1.0, flat-cache@1.2.1)
├── inquirer@0.12.0 (strip-ansi@3.0.1, ansi-regex@2.0.0, ansi-escapes@1.4.0, through@2.3.8, figures@1.7.0, cli-width@2.1.0, run-async@0.1.0, cli-cursor@1.0.2, readline2@1.0.1, string-width@1.0.2, rx-lite@3.1.2)
├── js-yaml@3.6.1 (esprima@2.7.3, argparse@1.0.9)
├── espree@3.3.2 (acorn@4.0.3, acorn-jsx@3.0.1)
├── table@3.8.3 (slice-ansi@0.0.4, ajv-keywords@1.1.1, string-width@2.0.0, ajv@4.8.2)
├── es6-map@0.1.4 (d@0.1.1, es6-symbol@3.1.0, event-emitter@0.3.4, es6-iterator@2.0.0, es6-set@0.1.4, es5-ext@0.10.12)
├── escope@3.6.0 (esrecurse@4.1.0, es6-weak-map@2.0.1)
└── lodash@4.16.4
```

[image]

ivankravets, 2016-10-25:
It's normal that it's not working. Please be careful. I mentioned above that source code of PlatformIO IDE should be unpacked/cloned to `~/.atom/packages/platformio-ide`. What is it `~/.atom/packages/platformio-ide/platformio-atom-ide`?

RebelCoderRU, 2016-10-25:
Sorry. That was my mistake. I was not careful. This worked, but there are more errors with Terminal not installing, and side bar not showing up. I use Qt Creator anyway and it works perfectly in it. I don't really need atom anymore, as you have helped to set up platformio anyway.

I think it is very rare case, that someone has the problem with very strict Proxy settings. So it is just that - proxy problems.

So I guess using pip with proxy fixed the thing and now I am using it with Qt.

ivankravets, 2016-10-25:
We have users with Proxy. However, they don't have problems with installation.

If Qt is cool for you, then I close this issue.

Happy Coding with PlatformIO! 😊

closed 2016-10-25 by ivankravets