platformio/platformio-core #814: Library dependencies not respecting platforms

ricardojlrufino, 2016-10-26:
### Configuration

**Operating system**: Ubuntu Linux 64

**PlatformIO Version**: PlatformIO, version 3.1.0

### Description of problem
I have a library that has dependencies on other libraries and some are specific per platform.

Previously I was managing the dependencies through a dependencies.h file in my library. The problem is that the user has to be editing the library files for each sketch, that's not cool.

I'm upgrading to platformio, and I am struggling to solve this problem, it seems to be a bug.

LIB: https://github.com/OpenDevice/opendevice-lib-arduino/tree/0.1.4

```json
  "dependencies": [
    {"name": "PubSubClient", "version": ">=2.6"},
    {"name": "EnableInterrupt", "version":  ">=0.9.5", "platforms" : "atmelavr"},
    {"name": "MFRC522", "version":  ">=1.1.8"},
    {"name": "Bridge", "version":  ">=1.6.1", "platforms" : "atmelavr"},
    {"name": "Ethernet", "version":  ">=1.1.2", "platforms" : "atmelavr"},
    {"name": "EEPROM", "version":  ">=2.0"},
    {"name": "SPI", "version":  ">=1.0"},
    {"name": "WiFi", "version":  ">=1.2.7", "platforms" : "atmelavr"},
    {"name": "ESP8266WiFi", "version":  ">=1.2.7", "platforms" : "espressif8266"}
  ],
```

#### Steps to Reproduce

_Starting an example_, I have two platforms set in my Skech: atmelavr / uno, espressif8266.
When I compile for UNO, throws an error that can not find the lib ESP8266WiFi that this being should only be "active" when compiling for the esp8266

### Actual Results (ERROR)

> pio run -e uno -vv

```
[Wed Oct 26 02:23:12 2016] Processing uno (platform: atmelavr, lib_deps: PubSubClient, lib_compat_mode: 2, board: uno, framework: arduino)
----------------------------------------------------------------------------------------------------------
Library Storage: /media/ricardo/Dados/TEMP/platformio-test/unotest/.piolibdeps
LibraryManager: Installing id=89
PubSubClient @ 2.6 is already installed
Collected 30 compatible libraries
Looking for dependencies...
Error: Could not find `ESP8266WiFi` dependency for `OpenDevice` library
======================================= [ERROR] Took 0.77 seconds =======================================

=============================================== [SUMMARY] ===============================================
Environment esp12e      [SKIP]
Environment uno [ERROR]
======================================= [ERROR] Took 0.77 seconds =======================================
```

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
env_default = uno

[common]
lib_deps = PubSubClient

[env:uno]
platform = atmelavr
board = uno
framework = arduino
lib_compat_mode = 2
lib_deps = ${common.lib_deps}
; enable auto-uploading
;targets = upload

[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
lib_compat_mode = 2
lib_deps = ${common.lib_deps}

```

**Source file to reproduce issue:**
```cpp
// I only use a blink and include:
#include <OpenDevice.h>
```

### Additional info
On lib,  i'm using: **"build" : { "libLDFMode" : 0 }**
On sketch: **lib_compat_mode = 2**

ivankravets, 2016-10-26:
Thanks a lot for the great report! This is a bug and I've just fixed it. Please re-test with the http://docs.platformio.org/en/stable/installation.html#development-version

ivankravets, 2016-10-26:
I found a problem with "EnableInterrupt" library
- http://platformio.org/lib/show/311/EnableInterrupt/manifest

This library doesn't has `platforms` field and PIO Build System can not check for platform compatibility. There are 2 options:

1. Male PR to https://github.com/GreyGnome/EnableInterrupt and specify `platforms` field.
2. Replace `{"name": "EnableInterrupt", "version":  ">=0.9.5", "platforms" : "atmelavr"},` with `{"name": "EnableInterrupt", "version":  ">=0.9.5"},`

closed 2016-10-26 by ivankravets

ricardojlrufino, 2016-10-26:
WoW... Thanks for the quick fix... 
It's working now finally .... was for more than four hours trying to solve ...

One question: Why dependencies are not automatically downloaded?

And a small problem: I think the verification platform is very narrow and some libraries are using the old name: "espressif
> .piolibdeps/MFRC522_ID63/library.json  

```json
{
  "name": "MFRC522",
  "frameworks": "arduino",
  "platforms": ["atmelavr", "ststm32", "teensy", "espressif"]
}
```

Running
>  pio run -e esp12e -v

I got:

> Processing esp12e (platform: espressif8266, lib_deps: PubSubClient, lib_compat_mode: 2, board: esp12e, framework: arduino)
> Library Storage: /media/ricardo/Dados/TEMP/platformio-test/unotest/.piolibdeps
> LibraryManager: Installing id=89
> PubSubClient @ 2.6 is already installed
> **Platform incompatible library /media/ricardo/Dados/TEMP/platformio-test/unotest/.piolibdeps/MFRC522_ID63**
> Platform incompatible library /media/ricardo/Dados/Programacao/platformio/packages/framework-arduinoespressif8266/libraries/SoftwareSerial
> Collected 24 compatible libraries
> More details about "Library Compatibility Mode": http://docs.platformio.org/en/stable/librarymanager/ldf.html#ldf-compat-mode
> Looking for dependencies...
> Error: Could not find `MFRC522` dependency for `OpenDevice` library

I did a pull request for these libraries, but I think it will happen to other ...

ivankravets, 2016-10-26:
> Why dependencies are not automatically downloaded?

It should work. If library is already installed, then PIO will not re-check "dependencies". Try to isntall
```
pio lib -g install https://github.com/OpenDevice/opendevice-lib-arduino/archive/0.1.4.zip
```

>  I think the verification platform is very narrow and some libraries are using the old name: "espressif

I'll think on it. Maybe, will add some hook for that.

ricardojlrufino, 2016-10-27:
> It should work. If library is already installed, then PIO will not re-check "dependencies". Try to isntall
I think it's because I'm using the lib directly without installing

Thanks

commit c2bf38f8 (2016-10-26): Improved handling of library dependencies specified in ``library.json`` manifest // Resolve #814