platformio/platformio-core #816: Can't pass -Wno flags for CI

FokkeZB, 2016-10-27:
What kind of issue is this?

- [X] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Mac OS 10.12

**PlatformIO Version** (`platformio --version`): 3.1.0

### Description of problem

I can't use `PLATFORMIO_BUILD_FLAGS` to pass `-Wno-parentheses-equality` to the compiler. It seems to receive it as `-no-parentheses-equality`?

#### Steps to Reproduce

1. Set env variable: `export PLATFORMIO_BUILD_FLAGS="-Wa,-Wno-parentheses-equality"`
2. Run CI: `platformio ci --lib=src --board=leonardo --board=sparkfun_promicro8 test`

### Actual Results

```
%HOME/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/as: unrecognized option `-no-parentheses-equality'
```

### Expected Results

Not to see this error and see it skip `parentheses-equality` instead.

### If problems with PlatformIO Build System:

**Source file to reproduce issue:**
```cpp
uint8_t ch = 10
if (ch == 10 || ch <= 100 && ch >= 80) {
}
```

ivankravets, 2016-10-27:
Why do you pass `"`? This is Unix and `export`. Try
```
export PLATFORMIO_BUILD_FLAGS=-Wa,-Wno-parentheses-equality
```

FokkeZB, 2016-10-27:
Mmm, because I followed http://docs.platformio.org/en/stable/envvars.html#envvar-PLATFORMIO_BUILD_FLAGS ?

Anyway, what you suggested works, thx!

closed 2016-10-27 by FokkeZB

ivankravets, 2016-10-27:
Thanks, fixed!

https://github.com/platformio/platformio/commit/3f955659a180e133a0d43fe0794ae36f89e7a5c4