platformio/platformio-core #817: Allow optional dependencies in library.json

ricardojlrufino, 2016-10-28:
- [x] Feature Request. 

I'm having some problems with dependencies management of my library (OpenDevice.io).

In my library I enable some features based on libraries (dependencies) used, this allows me to:
1. Prevent the user to download all dependencies manually
2. And leave a little smarter process.

I did it in the Arduino through the file (dependencies.h). But the inconvenience was the user having to change this file when using the examples. This is impractical.

The problems found in platformio:
- All library.json dependencies are mandatory
- The system needs to download all dependencies.
- If I use (lib_ignore) throws the error libraries not found

The improvement would be able to configure the library dependencies as **optional**.
And being able to "activate" them through libraries configured in the project.

To better understand how I do it just check includes the file:
https://github.com/OpenDevice/opendevice-lib-arduino/blob/master/src/OpenDevice.h

ivankravets, 2016-10-28:
> All library.json dependencies are mandatory

Why is it a problem? This is normal behaviour of each package manager where you specify libraries/packages on which project depends.

> The system needs to download all dependencies.

PIO Library Manager downloads only dependent libraries specified in manifest. If you don't use them in your project/library, they will not be included in the final firmware. GCC Linker will ignore them.

> If I use (lib_ignore) throws the error libraries not found

Please provide project where U can reproduce this issue

> The improvement would be able to configure the library dependencies as optional.

You do that with `lib_ignore` option.

ricardojlrufino, 2016-10-29:
I believe that will not be necessary, I am doing some more tests. I'm trying another implementation approach enable dynamic configuration.

ivankravets, 2016-10-29:
Please note that new PlatformIO Build System 3.0 follows by `#include <>` and understands  C/C++ Preprocessor conditional syntax (macros, etc). Please read
* http://docs.platformio.org/en/stable/librarymanager/ldf.html

ricardojlrufino, 2016-10-31:
Hi!

The problem was even with my implementation strategy. Really I have to have all the libs to compile. With PlatformIO it is easy because it downloads the dependencies automatically.

But with the user Arduino IDE will have to manually download .. = [

Anyway, thank you for your attention and congratulations for the project

closed 2016-10-31 by ricardojlrufino