platformio/platformio-core #819: Mbed double framework

Testato, 2016-10-28:
https://community.platformio.org/t/mbed-double-framework/962

sstaub, 2016-11-23:
With v5.2.3 mbed introduced macros for versioning in the mbed.h, maybe you
can use it in platformio.ini as an extra flag for the version (2 or 5) (e.g. mbedVersion = 2)
I found that there are following differences between v2 and v5:
- the standard api and rtos is the same for both versions
- v2 use the libraries net(Ethernet) and fs(filesystem) in /features/unsupported
- v5 use the libraries FEATURE_xxx, netsocket, nanostack, mbedtls ... in /feature/
The most important difference is that v5 use generally the rtos and the new libs need them
per design. There is a new event system(events) that should replace the classical timer lib.

Testato, 2016-11-23:
Exactly, the mbed-os use always the rtos, also for a simply blink.
This create a very big executable, and many mcu do not have flash space.

So the mbed-classic (v2) will alway mantained, and it is the base for the mbed-os.

For example i work on a Stm32f103 project, and i do not neet rtos feature, but whit his 64KBytes of flash i can only made a Blink if I use mbed-os

Testato, 2016-12-17:
Is there news on this feature ?

ivankravets, 2016-12-17:
@valeros has added support for mbed 5.2 a few days ago. See new branch
- https://github.com/platformio/builder-framework-mbed/tree/feature/new-mbed-builder
- https://github.com/platformio/platform-ststm32/tree/feature/new-mbed-builder

Nevertheless, mbed released 5.3 two days ago. We will re-update new builder and merge it to all development platforms where mbed framework is supported.

Testato, 2016-12-17:
thanks for the info,
but i mean if there is new on the Double Mbed separation core on Platformio, like explained  on my first platfomrio forum link 

> So for platformio it is a very important step create two framework.
> Framework mbed-dev
> Framework mbed-os
>

ivankravets, 2016-12-17:
@Testato This issue is on our Open Source TODO list. We will implement it as soon as possible.

5N44P, 2017-05-06:
Would be great to see this feature :)

ivankravets, 2017-06-22:
@valeros Could you explain here how to use mbed-sdk and mbed-os?

ivankravets, 2017-07-13:
Implemented in the latest  build script for mbed . See docs http://docs.platformio.org/en/latest/frameworks/mbed.html#configuration

Testato, 2017-07-14:
So without -DPIO_FRAMEWORK_MBED_RTOS_PRESENT we use mbed-classic ?

Why in the table of config option there isn't the -D ?
-DPIO_FRAMEWORK_MBED_RTOS_PRESENT vs PIO_FRAMEWORK_MBED_RTOS_PRESENT

Last question, so you never use mbed-os lib but mbed-classic + mbed-rtos lib ?

sstaub, 2017-07-14:
How to handle the networklibraries? They are different in mbed2 and mbed5, and some platforms does not have a network inteface. It would be helpfull to switch it on by a config flag.

valeros, 2017-07-14:
Hi @Testato!
`PIO_FRAMEWORK_MBED_RTOS_PRESENT` is just the name of the macro, so to add it to build flags you need to specify `-D` prefix for the compiler.
Is there any difference between `mbed-os` and `mbed-rtos`? I thought they are the same?

valeros, 2017-07-14:
Hi @sstaub!
`net` lib from features/unsupported no longer exists in the last framework version
Can you please provide an example that works in mbed online compiler with `mbed 2` and doesn't work in PlatformIO?

sstaub, 2017-07-14:
It seems that mbed 2 doesn't have the old network lib anymore.

Testato, 2017-07-14:
Hi @valeros and thanks for your effort on mbed-classic framework implementation

Mbed-classic use mbed-rtos for realtimeos functionality. mbed-rtos is not needed on mbed-OS because it is included.

So whitch core you actually use on platformio ?

The only way to work on mbed-classic and on mbed-os is create two framework like i already said.

If you compile a simple blink on mbed-classic (without mbed-rtos lib) you need only about 8kB of flash and 0.1kB of Ram.
Instead if you use mbed-os you always need for a blink about 21KB and 8kB of Ram.

This is because you cannot remove the rtos code also if you do jot need it.

Testato, 2017-07-14:
@ivankravets
May you leave this Issue open until we will test it ?
Personally i van test only next week, but if you added only a config option, not an entire second mbed framework, i'm agraid that it do not work

ivankravets, 2017-07-14:
@Testato Sure, please test and leave here all wishes. Thanks!

Testato, 2017-07-26:
This is my test on nucleo_f103rb

Blink Code
```
#include "mbed.h"
DigitalOut led1(LED1);
int main() {
    while (true) {
        led1 = !led1;
        wait(0.5);
    }
}
```

MBED-CLASSIC - Official online compiler:
Flash 20kB - Ram 1kB

MBED-OS - Official online compiler:
***main() runs in its own thread in the OS***
Flash 32kB - Ram 8kB

PLATFORMIO Without OS build flag
Flash 11kB - Ram 2kB

[env:nucleo_f103rb]
platform = ststm32
board = nucleo_f103rb
framework = mbed

PLATFORMIO With OS build flag
***main() runs in its own thread in the OS***
Error during compilation

[env:nucleo_f103rb]
platform = ststm32
board = nucleo_f103rb
framework = mbed
build_flags = -D PIO_FRAMEWORK_MBED_RTOS_PRESENT

Testato, 2017-07-26:
This is my test on nucleo_f103rb

Blink Code
```
#include "mbed.h"
DigitalOut led1(LED1);
int main() {
    while (true) {
        led1 = !led1;
        wait(0.5);
    }
}
```

**MBED-CLASSIC - Official online compiler:**
Flash 20kB - Ram 1kB

**MBED-OS - Official online compiler:**
_main() runs in its own thread in the OS_
Flash 32kB - Ram 8kB

**PLATFORMIO Without OS build flag**
Flash 11kB - Ram 2kB

[env:nucleo_f103rb]
platform = ststm32
board = nucleo_f103rb
framework = mbed

**PLATFORMIO With OS build flag AND #include "rtos.h"**
_main() runs in its own thread in the OS_
Flash 19kB - Ram 8kB

[env:nucleo_f103rb]
platform = ststm32
board = nucleo_f103rb
framework = mbed
build_flags = -D PIO_FRAMEWORK_MBED_RTOS_PRESENT

**CONCLUSION**
It seems to me that Platformio use only Mbed-Classic with rtos.h
It is not possible use directly the Mbed-OS library

Testato, 2017-07-26:
So on Platformio you use this:
https://developer.mbed.org/users/mbed_official/code/mbed/

Not this:
https://github.com/ARMmbed/mbed-os/

May you confirm ?

sstaub, 2017-07-26:
On [https://github.com/ARMmbed/mbed-os/releases](url) there are two different versions, 
in the moment mbed_lib_rev147 and mbed-os-5.5.3 I think platformIO use mbed_lib_revxxx.
mbed os5 use rtos, on mbed os2 rtos is an option.
At least we need a switch in platform.ini like "framework = mbed os2" and "framework = mbed os5"
to make a clear difference between the two versions.

valeros, 2017-07-27:
Hi, @sstaub!
As you can see both tags  `mbed_lib_rev147` and `mbed-os-5.5.3` point on the same commit `ed9d1da`, so there is no difference in source code between them.
Hi, @Testato!
We use stable releases from https://github.com/ARMmbed/mbed-os/
And about your research on firmware size, As I know the mbed online tool uses `ARMCC` compiler and we use `GCC`, so there might be a slight difference in binary sizes. Anyway, could you please point me where `mbed-os` is located in the mbed repository? I thought this part https://github.com/ARMmbed/mbed-os/tree/master/rtos is responsible for OS features?

sstaub, 2017-07-27:
@valeros you are right, it is the same code
I'm missing PIO_FRAMEWORK_MBED_NSAPI_PRESENT build flag for networking.
I think to use mbed os5 functions we need to set defines for the features
before including mbed.h , e.g.
`
#define MBED_CONF_RTOS_PRESENT
#include mbed.h
`
The old net library is gone and only available for online compiler (no export is
working), I will prepare a fork on github, be patient.

Testato, 2017-07-27:
mbed_lib_rev147 and mbed-os-5.5.3 is exactly the same code (i tested by a Diff), so it is here only for the name.
I suggest to use the name mbed-os-5.5.3 on Platformio because it is more clear.

Instead for Mbed-Classic it is here:
https://developer.mbed.org/users/mbed_official/code/mbed/

it is 236MB respect the 89MB of the mbed-os, so i think it is different :-)
There is also write that:
This is the mbed 2 library. If you'd like to learn about using mbed-os, please see the mbed-os docs.

hen we use online the Mbed-Classic, the online compiler use this code, and it is complete. If you want use rtos on this Classic framework you must manually add rtos.h library

ivankravets, 2019-05-23:
We support only mbed OS now. See detailed docs how to configure it => http://docs.platformio.org/en/latest/frameworks/mbed.html#configuration

See `bare-metal` profile without RTOS => https://github.com/ARMmbed/mbed-os/pull/9800

closed 2019-05-23 by ivankravets