platformio/platformio-core #820: Data loss on Import Arduino Project

rozling, 2016-10-30:
What kind of issue is this?

- [x ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [x ] Development Platform. All issues related to Development Platform should be reported to appropriate repository. Search it using link below
      https://github.com/platformio?query=platform-

- [ x] PlatformIO Core. If you’ve found a bug, please provide an information below.

Unknown / possibly any of the above

------------------------------------------------------------------

### Configuration

**Operating system**: macOS Sierra v10.12.1 

**PlatformIO Version** (`platformio --version`): IDE 1.6.0 | CLI 3.1.0

### Description of problem

Imported an existing sketch (modified from an example but saved in a new location) , via the 'Import Arduino Project' button.

This failed the first time, and I upgraded the Atom 'build' plugin to latest version.

Then I tried again and received error message: 
```PlatformIO: Failed to import an Arduino IDE Project!
Error: ENOENT: no such file or directory, scandir '/Users/xxx/Dropbox/Coding/Arduino/ESP Thing/Blynk_Sketch/Blynk_Sketch'```

#### Steps to Reproduce

1. Import Arduino Project - select folder.
2. Keep Project compatible between PlatformIO and Arduino IDE: true
3. Selected board: Sparkfun ESP8266 Thing <- 2x instances of this exist in the board list.

### Actual Results
- Error: ENOENT: no such file or directory, scandir '/Users/xxx/Dropbox/Coding/Arduino/ESP Thing/Blynk_Sketch/Blynk_Sketch'```
- on checking folder, Blynk_Sketch.ino is gone and the /src folder is empty.

### Expected Results
- Sketch should be imported
- All data should be left intact

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:thing]
platform=espressif8266
board=thing
framework=arduino

[platformio]
src_dir=Blynk_Sketch```

**Source file to reproduce issue:**
not available

### Additional info
n/a

rozling, 2016-10-30:
Managed to recover the code from /private/tmp/pio-arduino-import*************, in case anyone has the same issue.

ivankravets, 2016-10-30:
Please move this issue to https://github.com/platformio/platformio-atom-ide/issues

closed 2016-10-30 by ivankravets