platformio/platformio-core #821: Documentation bug: `upload_resetmethod` links are circular

spott, 2016-10-31:
When attempting to find out more about the `upload_resetmethod` for the ESP8266, I'm running into a circular reference problem:

[this](http://docs.platformio.org/en/stable/projectconf.html#upload-resetmethod) link, discusses the reset method, and references the ESP8266.  Clicking on that link, and searching the resulting page for `upload_resetmethod` sends you [here](http://docs.platformio.org/en/stable/platforms/espressif8266.html#custom-reset-method).  Unfortunately, the only information that this section gives you on the `upload_resetmethod` is by sending you back to the first link, which doesn't have any information.

I have no idea where the bugs for documentation go, so I'm making a bug report here.

Incidentally, I *still* don't know what `upload_resetmethod = ck` does.

ivankravets, 2016-10-31:
Thanks! See http://docs.platformio.org/en/latest/platforms/espressif8266.html#custom-reset-method

closed 2016-10-31 by ivankravets

spott, 2016-10-31:
You should reference here:

https://github.com/igrr/esptool-ck

Where they discuss what those resetmethods *mean* as well.  Thanks for the quick response!  I'm impressed.

ivankravets, 2016-10-31:
Done, thanks!

commit 92e3ca26 (2016-10-31): Improve docs for Custom Reset Method // Resolve #821