platformio/platformio-core #822: Espresso Lite V2.0 Error while uploading sketch

scotteon1, 2016-11-04:
While I try to upload sketch example _2000_ESPert_workshop which is from ESPert by Jimmy (current version 1.2.6). It shows following error. Even others kind of example sketch. What should i do?

Arduino: 1.6.12 (Mac OS X), Board: "ESPresso Lite 2.0, 80 MHz, 115200, 4M (3M SPIFFS), ck, Disabled, None"

Sketch uses 316,047 bytes (30%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 41,508 bytes (50%) of dynamic memory, leaving 40,412 bytes for local variables. Maximum is 81,920 bytes.
error: cannot access /dev/cu.usbmodem1411

error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.﻿

ivankravets, 2016-11-04:
Do you use PlatformIO IDE? http://platformio.org

scotteon1, 2016-11-05:
i am using Arduino Software (IDE) 1.6.12 to upload it and it shows the error that i had post previously.
if i use PlatformIO IDE to upload the sketch it shows ,
collect2: error: ld returned 1 exit status
*** [.pioenvs/espresso_lite_v2/firmware.elf] Error 1
so what is the problem?

ivankravets, 2016-11-14:
Place source code to `src` folder that contains `loop` and `setup` functions.

Please reopen if you still have problems.

closed 2016-11-14 by ivankravets

hidhir, 2017-01-18:
hye i have problem when i compile my code... i do not change anything with that code just copy n paste. but this error message get

**i need ur help...**

​Arduino: 1.8.1 (Windows 8.1), Board: "ESPresso Lite 2.0, 80 MHz, 115200, 4M (3M SPIFFS), ck, Disabled, None"
In file included from C:\Users\hidhir lutfi\Documents\Arduino\libraries\DHT_sensor_library\DHT_U.cpp:22:0:

C:\Users\hidhir lutfi\Documents\Arduino\libraries\DHT_sensor_library\DHT_U.h:25:29: fatal error: Adafruit_Sensor.h: No such file or directory

#include <Adafruit_Sensor.h>

^

compilation terminated.

Multiple libraries were found for "HttpClient.h"

Used: C:\Users\hidhir lutfi\Documents\Arduino\libraries\HttpClient

Not used: E:\Arduino\libraries\Bridge

exit status 1

Error compiling for board ESPresso Lite 2.0.

This report would have more information with

"Show verbose output during compilation"

option enabled in File -> Preferences.