platformio/platformio-core #823: Library Dependency Finder should take symlinked directories into account.

jasiek, 2016-11-04:
What kind of issue is this?

- [x] Feature Request. 

### Configuration

**Operating system**: All?

**PlatformIO Version** (`platformio --version`): 3.1.0

### Description of problem

I've got a project (https://github.com/londonhackspace/acnode-cl) which uses a vendored version of a particular library (https://github.com/londonhackspace/PN532). This repo has been vendored into `lib` within the project directory. As you can tell by browsing the repo above, it contains several directories, which contain the actual source code. I'm interested in using `PN532` and `PN532_HSU`. I have symlinked them both as relative links and checked that in - and several months later it appears to have ceased working.

I believe the behaviour should be that symlinked directories are indexed as containing source code when under the project-level `lib` directory, which will permit the build to proceed succesfully.

#### Steps to Reproduce

1. Check out the first repo above.
2. Try to built it.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:lptm4c1294ncpdt]
platform = titiva
framework = energia
board = lptm4c1294ncpdt
build_flags = !echo "-DLOGGING=1 -DGIT_REVISION='\"$(git rev-parse HEAD)\"'"
lib_deps =
  HttpClient
```

ivankravets, 2016-11-06:
How about http://docs.platformio.org/en/stable/projectconf.html#lib-extra-dirs ?

jasiek, 2016-11-06:
I'll try this tomorrow, but I really feel like symlinks should just be followed. I *think* that was the behaviour before as I've been able to build this project a few months ago.

ivankravets, 2016-11-06:
Please provide test project in archive with symlinks that demonstrates this issue.

jasiek, 2016-11-06:
It is right here: https://github.com/londonhackspace/acnode-cl

UntilSunrise, 2016-11-14:
I would appreciate symlink support, because i use an external private lib with the following directory structure:

|--lib
|  |--Bar  <- symlink
|  |  |--include
|  |     |- Bar.h
|  |  |--src
|  |     |- Bar.c

In my main.c i have to include Bar.h with a relative path because #include <Bar.h> cant be resolved and with the newest pio version i will get an "collect2: error: ld returned 1 exit status" error because the linker cant resolve the path to Bar.c. Platformio 2.11 did this all without an error.

ivankravets, 2016-12-01:
Could someone explain me how to reproduce this issue? I've jsut tested with latest PlatformIO 3.2b5
```
tree -L 2 /Volumes/MEDIA/tmp/pio3_test_projects/ldftest
/Volumes/MEDIA/tmp/pio3_test_projects/ldftest
├── asd
│   ├── Adafruit GFX Library_ID13
│   ├── Adafruit ILI9341_ID571
│   └── ILI9341_t3_ID540
├── lib
│   ├── Adafruit GFX Library_ID13 -> /Volumes/MEDIA/tmp/pio3_test_projects/ldftest/asd/Adafruit GFX Library_ID13/
│   ├── Adafruit ILI9341_ID571 -> /Volumes/MEDIA/tmp/pio3_test_projects/ldftest/asd/Adafruit ILI9341_ID571
│   └── readme.txt
├── platformio.ini
└── src
    └── main.cpp

8 directories, 3 files
```

Build output
```
Collected 34 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <SPI> v1.0 (/Users/ikravets/.platformio/packages/framework-arduinoespressif8266/libraries/SPI)
|-- <Adafruit ILI9341> v1.0.2 (/Volumes/MEDIA/tmp/pio3_test_projects/ldftest/asd/Adafruit ILI9341_ID571)
|   |-- <Adafruit GFX Library> v1.1.5 (/Volumes/MEDIA/tmp/pio3_test_projects/ldftest/asd/Adafruit GFX Library_ID13)
|   |-- <SPI> v1.0 (/Users/ikravets/.platformio/packages/framework-arduinoespressif8266/libraries/SPI)
xtensa-lx106-elf-size -B -d .pioenvs/d1_mini/firmware.elf
```

jasiek, 2016-12-02:
Can you provide a repo?

ivankravets, 2016-12-02:
Sure! See attached project.

[ldftest.tar.gz](https://github.com/platformio/platformio-core/files/626975/ldftest.tar.gz)

ivankravets, 2016-12-04:
Please reopen if you still have any problems

closed 2016-12-04 by ivankravets

trullock, 2021-07-30:
I have this problem on Windows 10.

I have a library (`mylib`) which contains `mylib.h` and `mylib.cpp`

`/lib/mylib` is a symbolic link (or junction, both fail) to a folder elsewhere

in my `main.cpp` i do: `#include "../lib/mylib/mylib.h"` and intellisense all works fine

When compiling, LDF doesn't pick up `mylib.cpp`, even if I include the folder explicity with `lib_extra_dirs`

If I make `lib/mylib` a real dir with the files in it, compilation works. Swap it out for a symlink, compilation fails because the `cpp` is never compiled so linking fails

Symlinks should be transparent to LDF, it shouldn't care if a folder is symlinked or not it should just explore it

trullock, 2021-08-02:
I'll reopen this as a new ticket as I suspect this will be lost