platformio/platformio-core #824: Environments as arguments

ivankravets, 2016-11-06:
Would be good to simplify commands that have `-e, --environments` options and allow to pass arguments as envs. For example,

* `pio run -e uno` is equal to `pio run uno`
* `pio run -e uno -e teeensy` is equal to `pio run uno teensy`
* `pio run -e release -t upload` => `pio run release -t upload`
* `pio test -e embedded` => `pio test embedded`
* `pio device monitor uno`, depends on @blackketter #787
*  etc...

jrobeson, 2017-01-18:
in that case, maybe it'd be nice to have a separate upload command also?
`pio upload env1 env2`, but error out if there's no upload target.

ivankravets, 2017-01-18:
@jrobeson I'm not sure that this issue will be implemented. It's just an idea.  According to `pio upload`,  we can't have this command because we have different type of upload commands: upload, program, uploadfs, etc.

ivankravets, 2018-06-02:
There is no support from the community for this feature and documentation already is well established using `-e, --environment`.

closed 2018-06-02 by ivankravets