platformio/platformio-core #829: Add option for using external tools

gaoyichuan, 2016-11-17:
- [x] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**:
Latest archlinux

**PlatformIO Version**:
3.2.0b2

Today readline on archlinux had been upgraded to version 7.0, but the avrdude binary bundled with platformio still use libreadline.so.6, which caused an error when uploading.
I think having an option for using external tools such as avrdude can be a good idea to solve this problem, since avrdude was upgraded together with readline.

If this option cannot be added, just upgrade the bundled avrdude can also be a choice. :)

ivankravets, 2016-11-22:
Could you try the latest `avrdude 6.3`? Just run `pio update`

ivankravets, 2016-11-22:
I've just checked under Arch Linux and it works.

closed 2016-11-22 by ivankravets

gaoyichuan, 2016-11-23:
Thanks, it works