platformio/platformio-core #830: Unable to include ssh user in lib_deps repository url

a2hill, 2016-11-18:
### Configuration

**Operating system**: macOS Seirra

**PlatformIO Version**: 3.1.0

### Description of problem
When including a git+ssh repository URL with the ssh username included the library installer will throw an AssertionError due to a version mismatch.

#### Steps to Reproduce

1. Add a lib_deps from a private repository (with username included): 
`git+ssh://git@gitlab.my-private-repo.com/a2hill/my-private-project.git#3.1.8`
2. Build the project
3. Receive error

### Actual Results
```
File "/Users/a2hill/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 211, in _install_from_url
    pkg_dir = self._install_from_tmp_dir(pkg_dir, requirements)
File "/Users/a2hill/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 249, in _install_from_tmp_dir
    assert tmpmanifest['version'] == requirements, mismatch_error
AssertionError: Package version 3.1.8 doesn't satisfy requirements gitlab.my-private-repo.com/a2hill/my-private-project.git#3.1.8
```

### Expected Results
Library should install correctly.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:teensy31]
platform = teensy
board = teensy31
framework = arduino
lib_deps =
  git+ssh://git@gitlab.my-private-repo.com/a2hill/my-private-project.git#3.1.8
```

**Source file to reproduce issue:**
```cpp
N/A
```

### Additional info
This is important for privately hosted repository solutions (like GitLab) where a single 'git' user is used to control access to the repositories via public/private key authentication.

I know that URL's from lib_deps are sent directly to the platformio lib install utility. So after looking at the documentation I saw an option for `platformio lib [STORAGE_OPTIONS] install <name>@<version>`. To me it looks like the url parser is splitting on the '@' symbol and treating the second token as a version when in reality it is just the rest of the URL.

sbrinkmann, 2017-05-21:
@ivankravets it seems to be that the issue returned with the latest version if pio core.

ivankravets, 2017-06-24:
Should be fixed in previous releases. See https://github.com/platformio/platformio-core/blob/develop/tests/test_managers.py#L127

commit 822ba634 (2016-12-02): Fix unable to include SSH user in ``lib_deps`` repository url // Resolve #830 Resolve #836
closed 2017-06-24 by ivankravets