platformio/platformio-core #831: Unable to authenticate with username and password for private repositories

a2hill, 2016-11-18:
### Configuration

**Operating system**: macOS Seirra

**PlatformIO Version**: 3.1.0

### Description of problem
When including a lib_deps dependency using a git+https repository URL where authentication is required the library installer will fail.

#### Steps to Reproduce

1. Add a lib_deps from a private repository: 
`git+https://gitlab.my-private-repo.com/a2hill/my-private-project.git#3.1.8`
2. Build the project
3. Receive error

### Actual Results
```
Cloning into '/Users/a2hill/projects/my-private-thing/.piolibdeps/installing-byV_0o-package'...
fatal: could not read Username for 'https://gitlab.my-private-repo.com': Device not configured

CalledProcessError: Command '['git', 'clone', '--recursive', '--depth', '1', '--branch', '3.1.8', 'https://gitlab.my-private-repo.com/a2hill/my-private-project.git#3.1.8', '/Users/a2hill/projects/my-private-thing/.piolibdeps/installing-byV_0o-package']' returned non-zero exit status 128
```

### Expected Results
Library installer should prompt for Username and Password

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:teensy31]
platform = teensy
board = teensy31
framework = arduino
lib_deps =
  git+https://gitlab.my-private-repo.com/a2hill/my-private-project.git#3.1.8
```

**Source file to reproduce issue:**
```
N/A
```

### Additional info
This is important for privately hosted repositories where authentication is required. Additionally if the username is included like `git+https://a2hill@gitlab.my-private-repo.com` the error will read `could not read Password for...`. You will also run into the issue #830 .

ivankravets, 2016-12-02:
What is your git version? My version asks me about username/password
```
pio run -d 
[Fri Dec  2 21:05:14 2016] Processing teensy31 (build_flags: -O  -D TEENSY31=1 -UUSB_SERIAL -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH -I./src, lib_deps: git+https://gitlab.com/user/project.git#0.6.0, board_f_cpu: 96000000, platform: teensy, board: teensy31, framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing project
git version 2.9.3 (Apple Git-75)
Cloning into '.../.piolibdeps/installing-E1qs8l-package'...
Username for 'https://gitlab.com':
...
```

ivankravets, 2016-12-04:
Please reopen if you still have any problems

closed 2016-12-04 by ivankravets

Zeppelin15, 2018-08-30:
I know this Issue is as old as time and has been closed two years ago but it is highly ranked in google, thus i posthere.

In my experience the following two commands does not give a password prompt with private repositories,:
`lib install
lib update
`
Th `pio run` command does work and asks for password. Thus using it you can initially install the library.
I don't know how to update the library once it is installed.

I I assume it works if you list the dependency as an git+ssh dependency under lib_deps, but have gotten around to testing

ivankravets, 2018-08-31:
@Zeppelin15 could you share a URI example of your deps?

Zeppelin15, 2018-09-03:
Listing a dependency to a private git  library in the form

`https://github.com/user/library.git`

prompts for a password with the run command.Get an authentication failure msg when trying `lib install`  or`lib update`

Both commands work when listing dependency as an ssl dependency, providing you setup your githubs ssl
`ssh://user@git.server.org/my-library`

Would be nice if the `lib` commands prompted for password instead of just failing

ivankravets, 2018-09-06:
@Zeppelin15, please file a new issue here https://github.com/platformio/platformio-core/issues and put a link to this issue.

tablatronix, 2020-05-11:
Just a note since I just had this on an old ini

changed
`platform = espressif8266/staging`
to
`platform = https://github.com/platformio/platform-espressif8266.git#feature/stage`