platformio/platformio-core #833: After upgrade to 3.1.0, unable to install 'atmelavr'

aphelps, 2016-11-20:
### Configuration

**Operating system**:  OSX 10.9.5

**PlatformIO Version** (`platformio --version`): 3.1.0

### Description of problem

I had a working older version of platformio and decided it was time to upgrade.  After running `sudo platformio upgrade` builds are now failing to download the 'atmelavr' framework:

```
amp@AMP-MBP-home:~/Dropbox/Arduino/HMTL/platformio/HMTL_Bringup [master]$ platformio run -t upload -e 1284
Please wait while upgrading PlatformIO ...
PlatformManager: Installing atmelavr
Warning! Package Mirror: 'X509' object has no attribute '_x509'
Looking for the another mirror...
Exception AttributeError: "'FileDownloader' object has no attribute '_request'" in <bound method FileDownloader.__del__ of <platformio.downloader.FileDownloader object at 0x10133ee50>> ignored
Error: Can not install 'atmelavr' with version requirements '*' for your system 'darwin_x86_64' 
```

(Running with 'sudo' gets the same result)

ivankravets, 2016-11-22:
Do you still have this problem?

aphelps, 2016-11-23:
I haven't checked since I filed this, I'll recheck it tonight.

aphelps, 2016-11-24:
Yes, this problem still exists.  I also just noticed that `platformio upgrade` also hits an error:

```
amp@AMP-MBP-home:~/Dropbox/Arduino/HMTL/platformio/HMTL_Bringup [master]$ platformio update
Please wait while upgrading PlatformIO ...
PlatformManager: Installing atmelavr
Warning! Package Mirror: 'X509' object has no attribute '_x509'
Looking for the another mirror...
Exception AttributeError: "'FileDownloader' object has no attribute '_request'" in <bound method FileDownloader.__del__ of <platformio.downloader.FileDownloader object at 0x102e9bfd0>> ignored
Error: Can not install 'atmelavr' with version requirements '*' for your system 'darwin_x86_64'
amp@AMP-MBP-home:~/Dropbox/Arduino/HMTL/platformio/HMTL_Bringup [master]$ platformio upgrade
Error: Can not retrieve the latest PlatformIO version
```

ivankravets, 2016-11-24:
Try to update Python Requests via `pip install -U requests`.

aphelps, 2016-11-24:
Looks like requests is up to date, version 2.12.1.  If its useful here's my currently installed python packages:

```
$pip freeze
altgraph==0.10.1
ansible==2.0.2.0
bdist-mpkg==0.4.4
BeautifulSoup==3.2.1
beautifulsoup4==4.3.2
blinker==1.3
bonjour-py==0.3
boto==2.36.0
bottle==0.12.10
cffi==1.6.0
click==5.1
colorama==0.3.7
configobj==4.7.2
cryptography==1.3.1
enum34==1.1.4
facebook-sdk==0.4.0
Flask==0.10.1
Flask-Login==0.2.9
Flask-Stormpath==0.4.4
Flask-WTF==0.12
-e git+git@github.com:aphelps/HTML.git@e4b8ec53890f0434aee87a25cbed4e82918b5e5e#egg=hmtl&subdirectory=python
httplib2==0.9.2
idna==2.1
ino==0.3.6
ipaddress==1.0.16
isodate==0.5.4
itsdangerous==0.24
Jinja2==2.7.2
kafka-python==0.9.3
lockfile==0.12.2
macholib==1.5
MarkupSafe==0.18
matplotlib==1.1.1
modulegraph==0.10.1
myo-python==0.2.2
numpy==1.6.2
oauth2client==1.2
oauthlib==1.0.3
ordereddict==1.1
paramiko==2.0.0
passlib==1.6.5
pbr==0.11.0
platformio==3.1.0
py==1.4.30
py2app==0.7.1
pyasn1==0.1.9
pycparser==2.14
pycrypto==2.6.1
PyDispatcher==2.0.5
PyJWT==1.4.0
pyobjc-core==2.3.2a0
pyobjc-framework-AddressBook==2.3.2a0
pyobjc-framework-AppleScriptKit==2.3.2a0
pyobjc-framework-AppleScriptObjC==2.3.2a0
pyobjc-framework-Automator==2.3.2a0
pyobjc-framework-CFNetwork==2.3.2a0
pyobjc-framework-Cocoa==2.3.2a0
pyobjc-framework-Collaboration==2.3.2a0
pyobjc-framework-CoreData==2.3.2a0
pyobjc-framework-CoreLocation==2.3.2a0
pyobjc-framework-CoreText==2.3.2a0
pyobjc-framework-DictionaryServices==2.3.2a0
pyobjc-framework-ExceptionHandling==2.3.2a0
pyobjc-framework-FSEvents==2.3.2a0
pyobjc-framework-InputMethodKit==2.3.2a0
pyobjc-framework-InstallerPlugins==2.3.2a0
pyobjc-framework-InstantMessage==2.3.2a0
pyobjc-framework-InterfaceBuilderKit==2.3.2a0
pyobjc-framework-LatentSemanticMapping==2.3.2a0
pyobjc-framework-LaunchServices==2.3.2a0
pyobjc-framework-Message==2.3.2a0
pyobjc-framework-OpenDirectory==2.3.2a0
pyobjc-framework-PreferencePanes==2.3.2a0
pyobjc-framework-PubSub==2.3.2a0
pyobjc-framework-QTKit==2.3.2a0
pyobjc-framework-Quartz==2.3.2a0
pyobjc-framework-ScreenSaver==2.3.2a0
pyobjc-framework-ScriptingBridge==2.3.2a0
pyobjc-framework-SearchKit==2.3.2a0
pyobjc-framework-ServiceManagement==2.3.2a0
pyobjc-framework-SyncServices==2.3.2a0
pyobjc-framework-SystemConfiguration==2.3.2a0
pyobjc-framework-WebKit==2.3.2a0
pyobjc-framework-XgridFoundation==2.3.2a0
pyOpenSSL==0.13
PyRSS2Gen==1.0.0
pyserial==3.2.1
pytest==2.7.2
pytest-runner==2.7
python-dateutil==2.5.2
pytz===2012d
PyYAML==3.11
pyzmq==14.2.0
requests==2.12.1
scipy==0.11.0
semantic-version==2.6.0
six==1.9.0
stormpath==2.1.1
Twisted==12.2.0
vboxapi==1.0
virtualenv==13.1.1
Werkzeug==0.9.4
WTForms==2.1
xattr==0.6.4
zope.interface==3.8.0
```

ivankravets, 2016-11-24:
Could you try to run it from Virtualenv?

aphelps, 2016-11-24:
Same result:

```
$ virtualenv venv
New python executable in venv/bin/python
Installing setuptools, pip, wheel...done.
$ source venv/bin/activate
$ platformio run -e 1284
Please wait while upgrading PlatformIO ...
PlatformManager: Installing atmelavr
Warning! Package Mirror: 'X509' object has no attribute '_x509'
Looking for the another mirror...
Exception AttributeError: "'FileDownloader' object has no attribute '_request'" in <bound method FileDownloader.__del__ of <platformio.downloader.FileDownloader object at 0x10ce79e50>> ignored
Error: Can not install 'atmelavr' with version requirements '*' for your system 'darwin_x86_64'
```

ivankravets, 2016-11-24:
Something is wrong with SSL on your machine. See http://stackoverflow.com/questions/40628315/python-requests-and-streaming-attributeerror-x509-object-has-no-attribute

aphelps, 2016-11-24:
Why would upgrading platformio bring out a problem if that's the issue?  It was building just fine prior to the upgrade.

Did v3 possibly make use of something that isn't standard on the older 10.9 OSX?  I believe I had v2.X prior to the upgrade.  I appear to have the most recent versions of both pyOpenSSL and requests library available.

ivankravets, 2016-11-24:
That is not a problem of PIO2 or PIO3. I don't have problem with PIO2 because we don't support it. As result, youhave obsolate packages and build sytem and there are no new packages for PIO2.

PIO3 is very stable and singnificantly improved branch. Please check [release history](https://github.com/platformio/platformio/releases) for critical bug fixes. As you can see, only the 1 release was issued after first PIO3. Now, 2 months withour releases. PIO 3.2 is coming soon and will finally drop support for PIO2. PIO2 was experimental project that had a lof of technical issues and problems according development platforms and library builder.

I highly recommect to migrate to PIO3. Here is a docs: http://docs.platformio.org/en/stable/migration.html

-------

Accroding your issue, please check this 
```
$ python
> import requests
> requests.get("https://dl.bintray.com/platformio/dl-platforms/manifest.json").json()
```

Do you see result?

aphelps, 2016-11-25:
Ok, it generates the `AttributeError: 'X509' object has no attribute '_x509'` error when using the system python, but gets results within a venv.  However I've installed platforio within that venv as well, and unfortunately still get that error.  However it is getting somewhat clear that this is some sort of versioning or incompatibility issue with my python installations, I'm going to take care of upgrading my OS and getting a clean python environment over the weekend and will see if that clears up the issue.

aphelps, 2016-11-26:
This particular problem went away after upgrading to OSX 10.12.1, so I'm going to close this issue.

However some of my projects no longer compile via platformio, with what looks like #include circular dependency issues.  I've been maintaining this code such that it compiled under the Ardiuno IDE as well as under platformio, did something change in regards to that with platformio 2.X and 3.X?  The projects still build fine from the current Ardiuno IDE.

I can refactor to resolve that circular dependency, but if there's a flag that can be set to resolve it I'd rather do that for the time being as I don't actually have much time to get this working before I need it in a couple days.

closed 2016-11-26 by aphelps

ivankravets, 2016-11-26:
> with what looks like #include circular dependency issues.

I made big improvemnts to LDF in PIO 3.2. Please try development version. Does it work?

aphelps, 2016-11-29:
I've updated to 3.2.0b2 and it looks like the circular dependency problem was eliminated, thanks!