platformio/platformio-core #834: #include ignored

nfored, 2016-11-22:
I have been using visual micro to write and upload my code, this has been working well. I wanted to be able to use your platform so I downloaded all the code and tried to build and its failing because its not including my libraries. 

------------------------------------------------------------------

### Configuration

RPI 3

PlatformIO, version 3.1.0

Include Section:
pi@aquapi:~/aquacontroller $ head -n15 src/AquaControl.ino

#include <Stepper.h>
#include <EEPROM.h>
#include <avr/wdt.h>
#include <RTClib.h>
#include <LiquidCrystal_I2C.h>
#include <LiquidCrystal.h>
#include <LCD.h>
#include <I2CIO.h>
#include <FastIO.h>
#include <OneWire.h>
#include <DallasTemperature.h>
#include <Wire.h>
#include <NewPing.h>
#include <DHT.h>

Here you can see I have copied the libraries to any possible location

pi@aquapi:~/aquacontroller $ ls src/
AquaControl.ino  DallasTemperature.h  debug.h  DHT.h  FastIO.h  I2CIO.h  LCD.h  LiquidCrystal.h  LiquidCrystal_I2C.h  NewPing.h  OneWire.h  RTClib.h  SI2CIO.h  SoftI2CMaster.h
pi@aquapi:~/aquacontroller $ ls lib/
DallasTemperature  debug  DHT  FastIO  I2CIO  LCD  LiquidCrystal  LiquidCrystal_I2C  NewPing  OneWire  readme.txt  RTClib  SI2CIO  SoftI2CMaster
pi@aquapi:~/aquacontroller $ ls  -R lib/
lib/:
DallasTemperature  debug  DHT  FastIO  I2CIO  LCD  LiquidCrystal  LiquidCrystal_I2C  NewPing  OneWire  readme.txt  RTClib  SI2CIO  SoftI2CMaster

lib/DallasTemperature:
DallasTemperature.h

lib/debug:
debug.h

lib/DHT:
DHT.h

lib/FastIO:
FastIO.h

lib/I2CIO:
I2CIO.h

lib/LCD:
LCD.h

lib/LiquidCrystal:
LiquidCrystal.h

lib/LiquidCrystal_I2C:
LiquidCrystal_I2C.h

lib/NewPing:
NewPing.h

lib/OneWire:
OneWire.h

lib/RTClib:
RTClib.h

lib/SI2CIO:
SI2CIO.h

lib/SoftI2CMaster:
SoftI2CMaster.h

Heres the Output As you can see it only loads 3 of the 15 libraries, because its not loading RTClib it doesn't have the overload for DateTime and fails to copy. Even if I remove that section of code which works when compiled on visual micro it starts failing it other locations again because of not having libs

avr-g++ -o "/home/pi/aquacontroller/src/AquaControl.ino.cpp" -x c++ -fpreprocessed -dD -E "/tmp/tmpxMPmna"
Collected 40 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Stepper> v1.1.3 (/home/pi/.platformio/packages/framework-arduinoavr/libraries/Stepper)
|-- <Wire> v1.0 (/home/pi/.platformio/packages/framework-arduinoavr/libraries/Wire)
|-- <EEPROM> v2.0 (/home/pi/.platformio/packages/framework-arduinoavr/libraries/EEPROM)
avr-g++ -o .pioenvs/megaatmega2560/src/AquaControl.ino.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=030100 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10612 -I/home/pi/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/pi/.platformio/packages/framework-arduinoavr/variants/mega -I/home/pi/.platformio/packages/framework-arduinoavr/libraries/Stepper/src -I/home/pi/.platformio/packages/framework-arduinoavr/libraries/Wire/src -I/home/pi/.platformio/packages/framework-arduinoavr/libraries/EEPROM/src -Isrc src/AquaControl.ino.cpp
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/PluggableUSB.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=030100 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10612 -I/home/pi/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/pi/.platformio/packages/framework-arduinoavr/variants/mega /home/pi/.platformio/packages/framework-arduinoavr/cores/arduino/PluggableUSB.cpp
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/Print.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=030100 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10612 -I/home/pi/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/pi/.platformio/packages/framework-arduinoavr/variants/mega /home/pi/.platformio/packages/framework-arduinoavr/cores/arduino/Print.cpp
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/Stream.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=030100 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10612 -I/home/pi/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/pi/.platformio/packages/framework-arduinoavr/variants/mega /home/pi/.platformio/packages/framework-arduinoavr/cores/arduino/Stream.cpp
/home/pi/aquacontroller/src/AquaControl.ino: In function 'void GetCurrentTime()':
/home/pi/aquacontroller/src/AquaControl.ino:397:27: error: 'class DateTime' has no member named 'dayOfWeek'

pi@aquapi:~/aquacontroller $ cat platformio.ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
lib_ldf_mode = 2

pi@aquapi:~/aquacontroller 

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

nfored, 2016-11-22:
https://github.com/nfored/AquaControl  Here is the full project and all the includes

ivankravets, 2016-11-22:
* What is it? https://github.com/nfored/AquaControl
* Why did you mix library headers with project source files? 

Please initialize new PlatformIO based project (`pio init -b megaatmega2560`), place source code to `src` direcotry and DON'T INSTALL/MODIFY libraries manually.

PlatformIO != Arduino != MS VS

We provide professional project development workflow. You can specify project dependencies and PlatformIO will do rest work automatically for you. See

*  http://platformio.org/lib

nfored, 2016-11-22:
I tried have only the .ino in the src directory and only the .h files in the lib directory. I appreciate your reply I will give that a try again.

Sent via the Samsung Galaxy Note5, an AT&T 4G LTE smartphone

  *   http://platformio.org/lib

-

nfored, 2016-11-22:
This is what I get if I only put the .ino in the src directory and do not manually load any .h files.

pi@aquapi:~/aquacontroller $ platformio run
[Tue Nov 22 13:26:02 2016] Processing megaatmega2560 (platform: atmelavr, board: megaatmega2560, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting AquaControl.ino
Collected 27 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Wire> v1.0
|-- <Stepper> v1.1.3
|-- <LiquidCrystal> v1.0.5
|-- <EEPROM> v2.0
Compiling .pioenvs/megaatmega2560/src/AquaControl.ino.o
Archiving .pioenvs/megaatmega2560/libFrameworkArduinoVariant.a
Indexing .pioenvs/megaatmega2560/libFrameworkArduinoVariant.a
Compiling .pioenvs/megaatmega2560/FrameworkArduino/CDC.o
Compiling .pioenvs/megaatmega2560/FrameworkArduino/HardwareSerial.o
/home/pi/aquacontroller/src/AquaControl.ino:5:20: fatal error: RTClib.h: No such file or directory
#include <RTClib.h>
^
compilation terminated.
Compiling .pioenvs/megaatmega2560/FrameworkArduino/HardwareSerial0.o
*** [.pioenvs/megaatmega2560/src/AquaControl.ino.o] Error 1

ivankravets, 2016-11-22:
I don't see dependencies in your project. Please read docs first amd "Project dependecies" section http://docs.platformio.org/en/stable/librarymanager/index.html

nfored, 2016-11-23:
Thank you so much.

Solution was not dependence but was to copy the whole lib directory not just .h files from my dev machine to a lib directory on my PI and then inside the platformio.ini give that directory to lib_dir, and then as you said only put the .ino file in the src and let your awesome program do the rest 

======================================================================================================= [SUCCESS] Took 39.77 seconds =======================================================================================================
pi@aquapi:~/aquacontroller $

closed 2016-11-23 by nfored