platformio/platformio-core #835: Mbed update for STM32

Testato, 2016-11-22:
on mbed version 129 is there an important update for the STM32 mcu.
May you update the platformio mbed version to 129 ?

ivankravets, 2016-11-22:
@valeros Please take a look at this PR https://github.com/platformio/builder-framework-mbed/pull/1 by @douardda. 

How is difficult to switch all development platforms to mbed 5.2? They stopped support 5.1 branch.

Thanks!

Testato, 2016-11-22:
If you want make a big update of all mbed framework, do not forgot the Double Framework issue.
https://github.com/platformio/platformio/issues/819
We should the two mbed framework, the mbed-classic and the mbed-os

Testato, 2016-11-22:
if you want to seriously review the entire Mbed Platformio management, do not forgot the 
https://github.com/platformio/platformio/issues/819 issue
Because it is important have the two official Mbed framewrok, the Mbed-Classic and the Mbed-Os

Split7fire, 2016-12-06:
@ivankravets May be it's wise to make a 2 separate mbed framework options? Mbed-classic and mbed-os.
Because everywhere on the site (simply) mbed mentioned.

ivankravets, 2016-12-06:
Fixed in https://github.com/platformio/platform-ststm32/commit/ed4151ea3d60f593f8cbc7b0f7a221a604345a4d

New development platform is planned for release tomorrow in pair with big update - PIO Core 3.2

ivankravets, 2016-12-06:
If you don't want to wait, see my comment 

* https://community.platformio.org/t/mbed-nucleo-f030r8-compilation-issue/1125/5?u=ivankravets

closed 2016-12-06 by ivankravets

Testato, 2016-12-06:
Why you close this issue ?
The update that i mean is present on version 129, not on 126

ivankravets, 2016-12-06:
@Testato We have this issue https://github.com/platformio/platformio-core/issues/819 where we will work on the new builder for mbed >=5.2.