platformio/platformio-core #837: LDF C++ parser error

blackketter, 2016-11-23:
I'm building platformio.ini file that targets two different architectures (ESP8266 and Teensy3) and need to include different sets of libraries. I'm using lib_deps and lib_ignore successfully, but the LDF C++ parser seems to be failing for me here. It is including both libraries in the build, though it should only include the first.

The code in the header file looks like this:

#if TEENSY31 == 1
#include <ILI9341_t3.h>
#else
#include <Adafruit_ILI9341.h>
#endif
and the platformio.ini flag looks like this:

build_flags = -O -D TEENSY31=1 -UUSB_SERIAL -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH -I./src
But the graph looks like this:

|-- <ILI9341_t3>
|   |-- <SPI> v1.0
...
|-- <Adafruit ILI9341> v1.0.0
|   |-- <Adafruit GFX Library> v1.1.5
|   |-- <SPI> v1.0
If I simply delete the line:

#include <Adafruit_ILI9341.h>
It builds fine.

Attached is a trivial example to reproduce:

[ldftest.zip](https://github.com/platformio/platformio/files/609151/ldftest.zip)

jrobeson, 2016-11-27:
Is this a regression, or not supported yet?

ivankravets, 2016-11-27:
I'll take a look on it later. Temporary solution is to `lib_ignore = LibName`.

ivankravets, 2016-11-28:
Please re-test with http://docs.platformio.org/en/latest/installation.html#development-version

ivankravets, 2016-11-28:
This is a problem of SCons engine. The SCons C/C++ Preprocessor currently is disabled. I refactored disabled code and fixed all issues. See my PR to SCons https://bitbucket.org/scons/scons/pull-requests/380/improvements-for-scons-c-c-pre-processor/diff

All these changes were applied to PlatformIO's `tool-scons` package with new version `3.20401.1`. 

Please report if it works.

blackketter, 2016-11-30:
Sorry, @ivankravets, still not working for me, alas.  The example doesn't find the appropriate libraries automatically and if I include them via lib_deps it automatically finds the wrong library, even though it's defined out..

ivankravets, 2016-11-30:
Please update http://docs.platformio.org/en/latest/installation.html#development-version

blackketter, 2016-11-30:
Apologies again @ivankravets, updated to the latest development version and we're a bit closer but the sample still requires the line:

`lib_deps = Adafruit ILI9341, Adafruit GFX Library`

under `[env:d1_mini]` to build.  I don't need any other lib_deps or lib_ignore directives.

ivankravets, 2016-11-30:
Thanks! I can't imagine how we are going to release this PlatformIO 3.2 with these new changes 😢

blackketter, 2016-11-30:
Dude, I'm really sorry.  I can make it work with lib_deps/lib_ignore so I wouldn't hold up a release for this issue.

ivankravets, 2016-11-30:
Hm... I've jsut re-tested again and all works very well!

1. Ok, I've just released PlatformIO 3.2 Beta 4. Please make `pio upgrade`
2. Check again.

ivankravets, 2016-11-30:
There 2 options:

* Install all dependencies globally `pio lib -g install`
* Use `lib_deps` and PIO will maintain deps directly for this project. **RECOMMENDED WAY**.

See my output where I have global deps:
```
[Wed Nov 30 19:27:43 2016] Processing teensy31 (platform: teensy, build_flags: -O  -D TEENSY31=1 -UUSB_SERIAL -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH -I./src, board: teensy31, board_f_cpu: 96000000, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 98 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <SPI> v1.0
|-- <ILI9341_t3>
|   |-- <SPI> v1.0
Linking .pioenvs/teensy31/firmware.elf
Building .pioenvs/teensy31/firmware.hex
Calculating size .pioenvs/teensy31/firmware.elf
text	   data	    bss	    dec	    hex	filename
25884	    704	   3864	  30452	   76f4	.pioenvs/teensy31/firmware.elf
================================================================================ [SUCCESS] Took 2.08 seconds ================================================================================

[Wed Nov 30 19:27:45 2016] Processing d1_mini (platform: espressif8266, board: d1_mini, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 36 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <SPI> v1.0
|-- <Adafruit ILI9341> v1.0.2
|   |-- <Adafruit GFX Library> v1.1.5
|   |-- <SPI> v1.0
Calculating size .pioenvs/d1_mini/firmware.elf
text	   data	    bss	    dec	    hex	filename
227020	   2248	  29464	 258732	  3f2ac	.pioenvs/d1_mini/firmware.elf
================================================================================ [SUCCESS] Took 1.41 seconds ================================================================================

========================================================================================= [SUMMARY] =========================================================================================
Environment teensy31	[SUCCESS]
Environment d1_mini 	[SUCCESS]
```

As you can see, no `lib_ignore` or other  hooks. All works very good.

P.S: This is forced me to detach from official SCons release. I've made [SCons PR #380](https://bitbucket.org/scons/scons/pull-requests/380/improvements-for-scons-c-c-pre-processor/diff) and don't know will it be merged and when. Nevertheles, PlatformIO has own package managment system where I can have own patched packages. I did the same with SCon. Now PIO 3.2 depends on patched SCons 2.5.1 and package version is `3.20501.1`.

Please check that you have `PlatformIO 3.2.0b4`.

ivankravets, 2016-11-30:
What is interesting, this huge refactoring to PIO LDF works 2-3x tiem faster then previous PlatformIO releases!!! 😊

blackketter, 2016-11-30:
Works as long as I have that single lib_deps line (I guess the auto-detect of some libraries isn't reliable.) 

I'll continue to use lib_deps as necessary. 

Keep up the AWESOME work!

closed 2016-11-30 by blackketter

ivankravets, 2016-11-30:
> Works as long as I have that single lib_deps line (I guess the auto-detect of some libraries isn't reliable.)

1. Please comment all lib deps in config
2. Please remove `.piolbdeps` from project
3. Install dependencies to global storage via `pio lib -g install "ILI9341_t3" "Adafruit ILI9341" "Adafruit GFX Library"`

Does it work now?

blackketter, 2016-11-30:
Yes, it works perfectly.  

I agree, lib_deps is preferred to global installs.  :)

What I'm surprised by is how ILI9341_t3 is automatically detected and installed but the Adafruit libraries are not. Best to explicitly specify libraries with lib_deps.

ivankravets, 2016-11-30:
> What I'm surprised by is how ILI9341_t3 is automatically detected and installed but the Adafruit libraries are not

What do you mean? Issue with `Adafruit GFX Library`?

blackketter, 2016-11-30:
AHA!  I see.   I was confused as to why the ILI9341_t3 library didn't need to be specified but the Adafruit ones did.

The ILI9341_t3 library is part of framework-arduinoteensy and is automatically available, though not visible via `pio lib list` 

Everything is working as expected now.

ivankravets, 2016-11-30:
How about to remove extra libraries from Arduino framework for Teensy?  I don't know why I added them. I propose to keep only basic libraries such as SPI, Wire, etc. WDYT?

blackketter, 2016-11-30:
The teensyduino installer from @PaulStoffregen includes all these libraries so teensy developers are used to having them all pre-installed and samples should "just work".

This bug was really about a problem with the LDF parser which is working awesome now.  I wouldn't mess with the framework-arduinoteensy unless there's a specific issue.

My confusion was around not knowing which libraries I had installed, since `pio lib [-g] list` didn't tell me about the ones that came with the framework.  That might be worthy of an enhancement

ivankravets, 2016-12-08:
PlatformIO 3.2 has been release. Please note that you need to change `lib_ldf_mode` to `chain+` to enable advanced C Preprocessor. See docs

* http://docs.platformio.org/en/stable/librarymanager/ldf.html#dependency-finder-mode

commit 8b121a1c (2016-11-28): Fix LDF C/C++ Preprocessor for conditional syntax // Resolve #837