platformio/platformio-core #838: St-Link update

Testato, 2016-11-25:
st-link is updated to a new 1.2.0 release, it include a year of changes and improvements
https://github.com/texane/stlink/releases/tag/1.2.0

ivankravets, 2016-11-25:
We updated our packages a few months ago. Please check `pio platform show ststm32`

Testato, 2016-11-25:
nope, it's the old 1.10 version

```
Package tool-stlink
-------------------
Type: uploader
Requirements: ~1.10200.0
Installed: Yes
Description: ST-Link
Url: https://github.com/texane/stlink
Version: 1.10200.1

```

ivankravets, 2016-11-25:
PlatformIO Package Version = X.Y.Z, where X & Z is our internal data and Y is real package version in format %2d%2d%%2d.

Testato, 2016-11-25:
Ok, thanks
I suggest to add a new row whit the original version format, example:
Package Version: 1.2.0
Platformio Version: 1.10200.1

commit acf56b28 (2016-12-02): Show vendor version of a package // Resolve #838
closed 2016-12-02 by ivankravets