platformio/platformio-core #843: Linker flags not passed to

dr-alun-moon, 2016-11-30:
PlatformIO Core. If you’ve found a bug, please provide an information below.

Running on Windows 10
PlatformIO version 3.1.0

I'm trying to write some pure assembly code for the Arduino Uno

The project can be found at https://github.com/dr-alun-moon/Arduino-Atom-Assember

Running the build through pio I get 

> [11/30/16 00:04:01] Processing uno (platform: atmelavr, board: uno, build_flags: --nostartfiles)
> -------------------------------------------------------------------------------
> Collected 0 compatible libraries
> Looking for dependencies...
> Project does not have dependencies
> avr-gcc -x assembler-with-cpp -g -Os -Wall -ffunction-sections -fdata-sections -flto 
> -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=030100 -DARDUINO_ARCH_AVR 
> -DARDUINO_AVR_UNO -Isrc -c -o .pioenvs\uno\src\blink.o src\blink.s
> avr-gcc -o .pioenvs\uno\firmware.elf -Os -mmcu=atmega328p -Wl,--gc-sections -flto 
> -fuse-linker-plugin .pioenvs\uno\src\blink.o -L.pioenvs\uno -Wl,--start-group -lm -Wl,--end-group
> c:/users/alun/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.9.2/../../../../avr/lib/avr5/crtatmega328p.o:(.init9+0x0): undefined reference to `main'
> collect2.exe: error: ld returned 1 exit status
> *** [.pioenvs\uno\firmware.elf] Error 1
> ========================== [ERROR] Took 2.94 seconds ==========================

Whereas If I run the avr-gcc directly
> avr-gcc -mmcu=atmega328p -nostartfiles blink.s

(my PATH has the .platformio version of avr-gcc)
I get a successful assembly without the reference to the missing main

could it be an issue with the `-fuse-linker-plugin`?

platformio.ini file follows
```ini
[env:uno]
platform = atmelavr
board = uno
build_flags: -nostartfiles
```

ivankravets, 2016-11-30:
Please use upper `.S`, I mean `blink.s`.

ivankravets, 2016-11-30:
> build_flags: -nostartfiles

Should be `build_flags = -Wl,-nostartfiles`

ivankravets, 2016-11-30:
The final solution:

#platformio.ini

```
[env:uno]
platform = atmelavr
board = uno
extra_script = extra_script.py
```

`extra_script.py` (place it near platformio.ini)
```py
Import("env")
env.Append(LINKFLAGS=["-nostartfiles"])
```

ivankravets, 2016-11-30:
Need to add this to PIO Examaples. Thanks!

dr-alun-moon, 2016-11-30:
Thanks for the solution.  I'm Curious why the flag never made it to gcc?  gcc accepts the -nostartfiles flag and does the right thing.  I'll have to dig into PlatforIO's code to see if there's a fix

ivankravets, 2016-11-30:
1. See docs for [build_flags](http://docs.platformio.org/en/stable/projectconf.html#build-flags). PlatformIO handles multiple options for the different scopes. `-Wl,-nostartfiles` is the right way to pass extra option to linker. However, GCC doesn't handle correctly this option. It requires directly `-nostartfiles` option without `-Wl`. You can re-test it. This is not a bug of PlatformIO.

2. PlatformIO proposes extra scripting to tune PlatformIO Build System On-The-Fly using its API. See `extra_script.py`.

I've just added your code to new example and updated `extra_script.py`. I tested it on my Uno and it works!. Thanks.

See https://github.com/platformio/platformio-examples/tree/develop/atmelavr-and-arduino/atmelavr-assember-blink

closed 2016-11-30 by ivankravets

pcordes, 2021-05-13:
Just for the record, since someone was misled by a previous comment (https://stackoverflow.com/questions/67516597/gcc-passing-nostartfiles-to-ld-via-gcc-for-minimal-binary-size)

> 1. See docs for [build_flags](http://docs.platformio.org/en/stable/projectconf.html#build-flags). PlatformIO handles multiple options for the different scopes. `-Wl,-nostartfiles` is the right way [...]

It's not that *GCC* doesn't handle it correctly, it's that `-nostartfiles` is a GCC option, not an `ld` option.   
If you passed it to `ld`, it would parse as `ld -n -o startfiles`.

The point of `-nostartfiles` is to leave out some files that GCC by default passes to `ld`.  `ld` only links what you tell it, so to leave stuff out you just doing pass it in the first place, not give it some option to ignore stuff.

commit 50df0bfc (2016-11-30): Add Assembler Blink example for AVR // Resolve #843