platformio/platformio-core #845: Forcing .hex generation on atmelsam platform

pelikhan, 2016-12-01:
We need to generate a ``.hex`` file for any board using the ``atmelsam`` platform. Looking at 

https://github.com/platformio/platform-atmelsam/blob/2c91782d8a758b8416d0a1debe384503fefaaf9b/builder/main.py#L259

it seems that using ``upload_protocol = stk500v2`` should be sufficient to force .hex generation. However, ``upload_protocol`` is ``sam-ba`` because it is loaded from the board definition:

https://github.com/platformio/platform-atmelsam/blob/2c91782d8a758b8416d0a1debe384503fefaaf9b/builder/main.py#L140

ivankravets, 2016-12-01:
Do you need HEX for other purpose or to use with PlatformIO for the custom `sam-ba`?

pelikhan, 2016-12-01:
We need the .hex file to do further processing on it. In general, a flag to force .hex generation would be best.

ivankravets, 2016-12-01:
You can force `upload_protocol = stk500v2` in the `platformio.ini` and then just `pio run`. It will generate HEX in `.pioenvs` directory.

pelikhan, 2016-12-01:
I tried that but the value is not used because of

https://github.com/platformio/platform-atmelsam/blob/2c91782d8a758b8416d0a1debe384503fefaaf9b/builder/main.py#L140

ivankravets, 2016-12-01:
Thanks, you are right. This is a bug. Please open an issue in apropriate repository.

* https://github.com/platformio/platform-atmelsam/issues

closed 2016-12-01 by ivankravets