platformio/platformio-core #848: Merging of gitignore files regenerates file regardless of actual content change

thomasf, 2016-12-06:
What kind of issue is this?

- [X] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

Ubuntu 16.10

**PlatformIO Version** (`platformio --version`):

Latest release and todays develop snapshot (PlatformIO, version 3.2.0b6)
 
### Description of problem

Even if the contents of .gitignore is the same the file is regenerated and possibly reordered.

#### Steps to Reproduce

before init:

```
# my assets
/ASSETS

# pio
.pioenvs
.clang_complete
.piolibdeps
```

### Actual Results

after init:
```
.pioenvs
.clang_complete
# my assets
/ASSETS
# pio
.piolibdeps
```

### Expected Results

```
# my assets
/ASSETS

# pio
.pioenvs
.clang_complete
.piolibdeps
```

### Additional info

I propose that the template gitignore doesnt make any structural changes to the
gitignore file if the functional conent is unchanged and that the template
renderers in general doesnt rewrite files that have the same content.

Also if a user for one reason of the other want's to override a line in
.gitignore by commenting it the gitignore merger should not readd it.

A sketchy test case data example:

```py

#  @staticmethod to be added into ProjectGenerator
def _merge_gitignore(base, contents):
    """_merge_gitignore returns a contents string if the merge results in any
changes. If a user has commented out a line it will not be readded by the merge.
If there are no changes the returned string is equal to the base argument."""
    result = []
    # ... code...
    return result

contents = '.pioenvs\n.clang_complete\n'

input1 = """# pio
.pioenvs
.clang_complete
.piolibdeps

# my assets
/ASSETS
"""
output1 = input1

input2 = """# pio
.pioenvs
.clang_complete
# .piolibdeps

# my assets
/ASSETS
"""
output2 = input2

input3 = """# pio
.pioenvs
# .piolibdeps

# my assets
/ASSETS
"""
output3 = """# pio
.pioenvs
# .piolibdeps

# my assets
/ASSETS
.clang_complete
"""

test_data = [
    (input1, output1),
    (input2, output2),
    (input3, output3),
]

# TODO: verify that input==output for all test_data

```

ivankravets, 2016-12-06:
Please `pio upgrade` and re-test. Thanks!

closed 2016-12-06 by ivankravets

thomasf, 2016-12-06:
Much better, the template files are still rewritten even if there aren't any changes though. 
This can cause annoying situations, for example emacs will complain that I need to reload the file before saving because it's last modified time has changed.  I guess that this can also cause problems in other similar scenarios.

btw. I figured out that I can add an en exclusion pattern in my `.gitignore` (`!.piolobdeps`)  to allow having whatever I want checked into my git-repository. Works for me.

ivankravets, 2016-12-06:
Thanks, fixed. Please 
```
pip install -U https://github.com/platformio/platformio/archive/develop.zip
```

ivankravets, 2016-12-06:
> btw. I figured out that I can add an en exclusion pattern in my .gitignore (!.piolobdeps) to allow having whatever I want checked into my git-repository

I don't understand why do you need that :) Is `lib_deps` bad? Or why you don't use `lib` folder?

thomasf, 2016-12-06:
Thanks, seems to work great. I guess this contents checking really should be
extended to all ide templates but it's no hurry. I think you should get on with
your release instead now.

As I understand it the moment I use another directory than what `libdeps_dir`
is set to the `platformio lib` command won't be able to update or install my
libs anymore? I just want to use the library manager while still retaining most
of the source code for building a project in the git history to have it
archived and browsable properly even if there is no internet connection
available at all.

ivankravets, 2016-12-07:
In this case, you can use `!.piolobdeps` as you described above.

Please note that you can have unlimited number of library storages. You can manage storage via
```
pio lib -d /path/to/custom/storage install myLib
```

Also, you can have custom folder in the project, for example `mylib`,
``` ini
[platformio]
lib_extra_dirs = mylib
```

And install some libs
```
pio lib -d mylib install OneWire
```

commit 22a27206 (2016-12-06): Fix merging of ".gitignore" files when re-initialize project // Resolve #848