platformio/platformio-core #850: Produce less noisy output when -s/silent arguments are used

thomasf, 2016-12-07:
* Don't print anything that isn't a warning or error when -s is used
* The `init` command doesnt have a `--silent` argument.

I started something here https://github.com/platformio/platformio-core/pull/849 but aborted because I'm too tired.. 

It would probably be a good thing to utilize some simple leveled logging system for all status output functions..  Either implement a small one directly in pio-core or make use of pythons build in `logging` module as long as there are no integration problems with `click`

closed 2016-12-08 by ivankravets

thomasf, 2016-12-08:
The "please wait.." line is also useless for a silent output.

ivankravets, 2016-12-08:
> The "please wait.." line is also useless for a silent output.

Where do you see that?

thomasf, 2016-12-08:
Sorry, I accidentally reset to the master branch instead of develop. All good now!

commit ee086ff5 (2016-12-08): Produce less noisy output when “-s/—silent” options are used for run and init commands // Resolve #850