platformio/platformio-core #851: Platformio-ide no install windows 7

tistructor, 2016-12-08:
- [ ] PlatformIO Core. 

------------------------------------------------------------------

### Configuration

**Operating system**: windows 7 Professional

**PlatformIO Version** (`platformio --version`): don't work

**IDE Version**  1.7.1

### Description of problem
Platformio-IDE Core Failed to Retrieve.
I tried everything but the problem persists.
I followed the installation guide and everything seems to be fine but when I start Atom, is the welcome screen of platformio-ide but can not download or to start the core.
performing pip in the end the result is this:
<
PS C:\Users\fdebernardi> pio
Traceback (most recent call last):
  File "C:\Python27\Lib\runpy.py", line 174, in _run_module_as_main

    "__main__", fname, loader, pkg_name)
  File "C:\Python27\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Users\fdebernardi\.atom\packages\platformio-ide\penv\Scripts\pio.exe\__main__.py", line 5, in <module>
  File "c:\users\fdebernardi\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\__main__.py", line 24, in <module>
    from platformio import __version__, exception, maintenance
  File "c:\users\fdebernardi\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\maintenance.py", line 25, in <module>
    from platformio.commands.lib import lib_update as cmd_lib_update
  File "c:\users\fdebernardi\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\commands\lib.py", line 32, in <module>
    " library storage `%s`" % join(util.get_home_dir(), "lib"))
  File "c:\users\fdebernardi\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\util.py", line 214, in get_home_dir
    os.makedirs(home_dir)
  File "c:\users\fdebernardi\.atom\packages\platformio-ide\penv\lib\os.py", line 150, in makedirs
    makedirs(head, mode)
  File "c:\users\fdebernardi\.atom\packages\platformio-ide\penv\lib\os.py", line 150, in makedirs
    makedirs(head, mode)
  File "c:\users\fdebernardi\.atom\packages\platformio-ide\penv\lib\os.py", line 150, in makedirs
    makedirs(head, mode)
  File "c:\users\fdebernardi\.atom\packages\platformio-ide\penv\lib\os.py", line 157, in makedirs
    mkdir(name, mode)
WindowsError: [Error 3] Impossibile trovare il percorso specificato: 'Q:\\\\ '
PS C:\Users\fdebernardi>
>

#### Steps to Reproduce

For each new installation before doing a file cleaning eliminating atom from the control panel and folders in appdata and .atom

### Actual Results
Platformio-IDE CORE not installed.

### Additional info
I installed platformio-ide on ubuntu 15.04 mate smoothly.
On another PC with windows7 U. it worked the first time and on another laptop
windows7 with U. I had to struggle after various installations and removals, but is now working.
Only the latter laptop with windows 7 professional does not want platformio.

Thanks
Fabio

ivankravets, 2016-12-08:
1. Close Atom
2. Remove manually this folder `c:\users\fdebernardi\.atom\packages\platformio-ide\penv`
3. Replace this file `c:\users\fdebernardi\.atom\packages\platformio-ide\lib\install\command.js` with [this](https://raw.githubusercontent.com/platformio/platformio-atom-ide/develop/lib/install/command.js).
4. Start Atom

tistructor, 2016-12-08:
Same problem. Even after restart laptop.

I create a new user and try the new solution.
I already did this attempt the error was different just were not copied pio.exe and platformio.exe files in the folder.

ivankravets, 2016-12-08:
@tistructor have you tried this solution https://github.com/platformio/platformio-core/issues/851#issuecomment-265729769 ?

tistructor, 2016-12-08:
Now I try.

I clean create the user and install again. 15min

I do not understand why in the installation programs under control panel gives me atom installed 08/03/2016 I installed a few days. !
Maybe it has nothing to do.

ivankravets, 2016-12-08:
> 08/03/2016 I installed a few days. !

Try the latest Atom from https://atom.io

tistructor, 2016-12-08:
Same problem.  
platformio installed as a package by atom 1.12.6 just installed in the new user.
Failed to retrieve.

tistructor, 2016-12-08:
and replace command.js with new file.  
Same problem.

tistructor, 2016-12-08:
In my pc installed python 2.7.12 (64bit) and LLVM 3.9.0 64bit

ivankravets, 2016-12-08:
@tistructor please contact me@ikravets.com

tistructor, 2016-12-08:
My English is very bad use google translator.
No problem?

ivankravets, 2016-12-08:
@tistructor sure, I understand you, please mail me

tistructor, 2016-12-08:
email sent.

tistructor, 2016-12-08:
Thank you.
A great job.
I also arranged creo 3d, now also retrieves information without setting the environment variable HOME.
Problem solved.

;_)

ivankravets, 2016-12-08:
Thank you for debug session!

ivankravets, 2016-12-08:
The problem was with custom `%HOME%` and invalid directory

closed 2016-12-08 by ivankravets

tistructor, 2016-12-08:
It 'was fun.
;-)

tistructor, 2016-12-08:
I had already informed the technical create crazy installation of their software. !

aufdenkampe, 2017-02-08:
@ivankravets , can you provide additional details about how to fix the path?

I also get the `Platformio-IDE Core Failed to Retrieve` error, but I'm running on MacOS 10.12.3 with an Anaconda Python installation.

I initially couldn't install the package in Atom, with this error: `PlatformIO: Unable to create a virtualenv`

Then, after much messing with paths, assuming it was an Anaconda thing, I *think* I discovered that it wasn't installed anywhere, so I did ran these two commands in the terminal:
```
    x$ pip install virtualenv
    x$ virtualenv ~/.atom/packages/platformio-ide/penv
```
for the second command, it didn't seem to fully execute, and I got an error with this at the end (full error below):

   `OSError: Command /Users/aaufdenkampe/...-ide/penv/bin/python - setuptools pip wheel failed with error code 1`

Despite that, it allowed the PlatformIO packages to install, but in the PlatformIO tab of Atom, I get this:

    `Core: Failed to Retrieve`

aufdenkampe, 2017-02-09:
@ivankravets , here is the full error.
```
ExcaliburMac:~ aaufdenkampe$ virtualenv ~/.atom/packages/platformio-ide/penv
New python executable in /Users/aaufdenkampe/.atom/packages/platformio-ide/penv/bin/python
Installing setuptools, pip, wheel...
  Complete output from command /Users/aaufdenkampe/...-ide/penv/bin/python - setuptools pip wheel:
  Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "//anaconda/lib/python2.7/tempfile.py", line 32, in <module>
    import io as _io
  File "//anaconda/lib/python2.7/io.py", line 51, in <module>
    import _io
ImportError: dlopen(/Users/aaufdenkampe/.atom/packages/platformio-ide/penv/lib/python2.7/lib-dynload/_io.so, 2): Symbol not found: __PyCodecInfo_GetIncrementalDecoder
  Referenced from: /Users/aaufdenkampe/.atom/packages/platformio-ide/penv/lib/python2.7/lib-dynload/_io.so
  Expected in: flat namespace
in /Users/aaufdenkampe/.atom/packages/platformio-ide/penv/lib/python2.7/lib-dynload/_io.so
----------------------------------------
...Installing setuptools, pip, wheel...done.
Traceback (most recent call last):
  File "//anaconda/bin/virtualenv", line 11, in <module>
    sys.exit(main())
  File "//anaconda/lib/python2.7/site-packages/virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "//anaconda/lib/python2.7/site-packages/virtualenv.py", line 945, in create_environment
    download=download,
  File "//anaconda/lib/python2.7/site-packages/virtualenv.py", line 901, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "//anaconda/lib/python2.7/site-packages/virtualenv.py", line 797, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /Users/aaufdenkampe/...-ide/penv/bin/python - setuptools pip wheel failed with error code 1
```

aufdenkampe, 2017-02-13:
I finally got the install to complete by following the advice on how to link the env installed by conda, given here: https://github.com/platformio/platformio-atom-ide/issues/152#issuecomment-239866260

However, this led to a "Core: Failed to retrieve" error, which I'm following up on here: https://github.com/platformio/platformio-atom-ide/issues/218