platformio/platformio-core #852: Windows 7 - Upgraded Platform IO and I doesn't build

thehellmaker, 2016-12-08:
Upgraded platform io and I get the following error while building. I upgraded platform io using pip.Tha didn't help either

Everything Atom can do is in the Command Palette. See it by using Ctrl+Shift+P
1.9 s
platformio run --target upload
[12/08/16 09:45:55] Processing nodemcuv2 (platform: espressif8266, build_flags: -lstdc++ -lsupc++, board: nodemcuv2, framework: arduino)
 
Error: Traceback (most recent call last):
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\__main__.py", line 106, in main
    cli(None, None, None)
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 873, in invoke

    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\commands\run.py", line 109, in cli
    results.append((envname, ep.process()))
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\commands\run.py", line 172, in process
    result = self._run()
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\commands\run.py", line 266, in _run
    return p.run(build_vars, build_targets, self.silent, self.verbose)
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\platform.py", line 261, in run
    variables['build_script'] = self.get_build_script()
  File "c:\users\thehellmaker\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\platform.py", line 384, in get_build_script
    raise NotImplementedError()
NotImplementedError
 
============================================================
 
An unexpected error occurred. Further steps:
 
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command
 
* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html
 
* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues
 
============================================================

ivankravets, 2016-12-08:
Please open `Menu: PlatformIO > Terminal` and type this command
```
pio platform uninstall espressif8266
```

Build project again

ivankravets, 2016-12-09:
Please reopen if you still need a help

closed 2016-12-09 by ivankravets