platformio/platformio-core #853: Unable to Init and Build projects after upgrade to 3.2.1

johnnyegel, 2016-12-12:
After upgrading to platformio v3.2.1 I'm no longer able to build my Arduino Micro project. I've tried re-installing plaformio and tried the development version, but the problem stays the same. I also tried to re-initialize the project, which fails exactly with the same error message. My python version is 2.7.10.

Here is the output from the project init:
```
% pio init -b micro --ide eclipse                                                                                                                                                                          !4539

The current working directory XXX will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories have been created in /home/johnny/dev/git/projects/magnetstirrer
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries
PlatformManager: Installing atmelavr
Warning! Package Mirror: 'Extensions' object has no attribute 'get_extension_for_class'
Looking for other mirror...
Exception AttributeError: "'FileDownloader' object has no attribute '_request'" in <bound method FileDownloader.__del__ of <platformio.downloader.FileDownloader object at 0x7f8774d148d0>> ignored
Error: Can not install 'atmelavr' with version requirements '*' for your system 'linux_x86_64'
```

When running "pio run" before I deleted the platformio.ini and .pioenvs folder, the output was the same:
```
PlatformManager: Installing atmelavr
Warning! Package Mirror: 'Extensions' object has no attribute 'get_extension_for_class'
Looking for other mirror...
Exception AttributeError: "'FileDownloader' object has no attribute '_request'" in <bound method FileDownloader.__del__ of <platformio.downloader.FileDownloader object at 0x7f8774d148d0>> ignored
Error: Can not install 'atmelavr' with version requirements '*' for your system 'linux_x86_64'
```

The content of my platformio.ini file is now empty, likely due to me trying to re-init it. However, it was stock before, with only the defaults from an init in it.

ivankravets, 2016-12-13:
The problem is linked with https://github.com/kennethreitz/requests

Do you have multiple `requests` in the system? Could you share here an output of `pip list` command?

Also, try previous version of requests: `pip install "requests<2.12"`.

johnnyegel, 2016-12-13:
Thanks...  `pip install "requests<2.12"` worked perfectly :-)

It seems the problem was version 2.12.3 which was installed before. Now with version 2.11.1, it works as expected.

closed 2016-12-13 by ivankravets