platformio/platformio-core #854: Programming custom board using SWD (NRF51822)

krystiansliwa, 2016-12-13:
Hi, 
I bought some time ago this interesting board:
[image]

It has I2C accelerometer (LIS3DH) and is based on NRF51822 chip.
As you can see board has SWD interface for programming/debugging. I tested it and successfully compiled embed blink program. Also I write hex file to it using OpenOCD and STLinkV2 programmer.

But here is my question:
**It is possible to compile and upload hex file to this board using platformio tools with SWD and STLinkv2**?

This is my current config:
`
platform = nordicnrf51
framework = mbed
board = redBearLab
`

I can write for example pre and post action scripts for that:

1. Run OpenOCD interface for nrf51 and Stlink_v2
2. Connect to debugger using telnet
3. Connect to board and halt it
4. Write image using generated hex file
5. Reset board
6. Disconnect from debugger
7. Close debugger

But it is complicated and I am looking for other better solution.

ivankravets, 2016-12-13:
1. Does this board have a name?
2. Who is a vendor?
3. You can modify this script https://github.com/platformio/platform-nordicnrf51/blob/develop/builder/main.py

This file is located in `~/.platformio/platforms/platform-nordicnrf51/builder`.

krystiansliwa, 2016-12-14:
I think this board don't have any name, and vendor is unknown. You can find this board for example here: [http://www.ebay.com/itm/1PCS-NEW-Nrf51822-LIS3DH-Bluetooth-Module-Development-board/201630391054?_trksid=p2047675.c100005.m1851&_trkparms=aid%3D222007%26algo%3DSIC.MBE%26ao%3D2%26asc%3D38530%26meid%3De127d261990949379bbf0e4957d4d614%26pid%3D100005%26rk%3D2%26rkt%3D6%26sd%3D222196941463](url)

How to modify it? Where I should start? 
Now I have created new custom board but I don't know how to change programmer settings.

ivankravets, 2016-12-14:
1. Please move this issue to https://github.com/platformio/platform-nordicnrf51/issues
2. Here is source code of Nordic nRF51 development platform https://github.com/platformio/platform-nordicnrf51

closed 2016-12-14 by ivankravets

by2coffee, 2016-12-14:
@krystiansliwa If you are going to try this, I am ready to test (with a different board, but upload over SWD should be same)

I think you should start here https://github.com/platformio/platform-nordicnrf51/blob/a4a331c68ed9e7e169d1c80c999ba23c5ca31565/builder/main.py#L128

krystiansliwa, 2016-12-15:
@by2coffee During weekend I will have some time to check this and let you know about results. Also move this issue to proper location