platformio/platformio-core #856: Remove line-buffering from programmer

eric-wieser, 2016-12-21:
What kind of issue is this?

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): 3.2.1

### Description of problem

When the program run by `platformio run --target upload` emits progressbars, these are linebuffered, and therefore only the final result is visible

#### Steps to Reproduce

1. `platformio init --board mega_pic32`
2. Write some basic code
3. `platformio run --target upload`

### Actual Results

Progress bars are emitted once complete, and not during execution. Upload is still successfull
```
Uploading .pioenvs\mega_pic32\firmware.hex
Programmer for Microchip PIC32 microcontrollers, Version 2.1.24
Copyright: (C) 2011-2015 Serge Vakulenko
Adapter: STK500v2 Bootloader
Program area: 1d000000-1d1fffff
Processor: Bootloader
Flash memory: 2048 kbytes
Boot memory: 80 kbytes
Data: 56364 bytes
Erase: done
Program flash: ########################################################## done
Verify flash: ######################################################### done
Program rate: 2048 bytes per second
```
### Expected Results

Output is duplicated through platformio character-by-character

eric-wieser, 2016-12-21:
Hmm, this actually seems to be
* [not part of code](https://github.com/platformio/platform-microchippic32/blob/develop/builder/main.py#L160-L164)
* Possible part of `scons` instead

eric-wieser, 2016-12-21:
Nope, [this is `core` after all](https://github.com/platformio/platformio-core/blob/403da8e22b86e139fbb87ebe6bdddfd5b1bfbc6c/platformio/managers/platform.py#L285-L309)

ivankravets, 2016-12-21:
This issue is related to https://github.com/platformio/platformio-core/blob/develop/platformio/util.py#L87

I played with it in a previous year. I don't remember why I decided to save it without changes.

I mark it as a feature.

eric-wieser, 2016-12-21:
Seems to me that parsing output for error messages, yet allowing rewritable output, are behaviours that are sort of at odds with each other, and maybe need some platform-specific config to toggle

eric-wieser, 2016-12-21:
Unless we did something unusual like printing `\r` and repeating the line in red if enough characters for an error are found

commit f844d9cb (2019-05-17): Remove line-buffering from "platformio run" command which was leading to omitting progress bar from upload tools // Resolve #856, Resolve #857
closed 2019-05-17 by ivankravets