platformio/platformio-core #861: Cosa library integration

ilovesushi, 2016-12-25:
https://github.com/mikaelpatel/Cosa

Is there any way of integration of this great library to plATFORMIO, 
OR simply adding path to project file to  automatically include base and additional libraries from external folders

It simply integrates with arduino just by copying the library folder, Arduino automaticly tracks all dependency, but with platformio i have to include every additional piece of code, 

[env:custom_lib_dirs]
lib_extra_dirs = c:\Downloads\Cosa\cores\cosa\

doesnt help at all, no including
-----------------------
`"C:\Program Files (x86)\JetBrains\CLion 2016.3\bin\cmake\bin\cmake.exe" --build C:\Downloads\Clion_project\cosa2\cmake-build-debug --target PLATFORMIO_BUILD -- -j 8
[12/25/16 05:10:09] Processing pro8MHzatmega168 (platform: atmelavr, board: pro8MHzatmega168, framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
Collected 27 compatible libraries
Looking for dependencies...
Project does not have dependencies
Compiling .pioenvs\pro8MHzatmega168\src\main.o
src\main.cpp:27:52: fatal error: c:\Downloads\Cosa\cores\cosa\cosa\Cosa.h: No such file or directory
#include "c:\Downloads\Cosa\cores\cosa\cosa\Cosa.h"
^
compilation terminated.
Compiling .pioenvs\pro8MHzatmega168\FrameworkArduino\wiring.o
Compiling .pioenvs\pro8MHzatmega168\FrameworkArduino\wiring_analog.o
Compiling .pioenvs\pro8MHzatmega168\FrameworkArduino\wiring_digital.o
Compiling .pioenvs\pro8MHzatmega168\FrameworkArduino\wiring_pulse.o
Compiling .pioenvs\pro8MHzatmega168\FrameworkArduino\wiring_shift.o
*** [.pioenvs\pro8MHzatmega168\src\main.o] Error 1

[12/25/16 05:10:10] Processing custom_lib_dirs (lib_extra_dirs: c:\Downloads\Cosa\cores\cosa\)

 [ERROR] Took 1.41 seconds 
Error: Please specify platform for 'custom_lib_dirs' environment
mingw32-make.exe[3]: *** [CMakeFiles/PLATFORMIO_BUILD] Error 1
CMakeFiles\PLATFORMIO_BUILD.dir\build.make:56: recipe for target 'CMakeFiles/PLATFORMIO_BUILD' failed
CMakeFiles\Makefile2:103: recipe for target 'CMakeFiles/PLATFORMIO_BUILD.dir/all' failed
mingw32-make.exe[2]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/all] Error 2
mingw32-make.exe[1]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/rule] Error 2
mingw32-make.exe: *** [PLATFORMIO_BUILD] Error 2
CMakeFiles\Makefile2:115: recipe for target 'CMakeFiles/PLATFORMIO_BUILD.dir/rule' failed
Makefile:130: recipe for target 'PLATFORMIO_BUILD' failed
`

ivankravets, 2016-12-25:
@mikaelpatel could you help us here? 

* http://docs.platformio.org/en/latest/platforms/custom_platform_and_board.html
* Frameworks for AVR https://github.com/platformio/platform-atmelavr/tree/develop/builder/frameworks

mikaelpatel, 2016-12-25:
@ivankravets Please see https://github.com/mikaelpatel/Cosa/tree/master/build/PlatformIO, this work was contributed by Sinisha Djukic. Please note that this is only an initial step and only a limited number of boards are supported. Also there are a lot of library properties files missing.