platformio/platformio-core #862: Unknown board ID 'uno'

bubnovKelnik, 2016-12-27:
Just updated platformio CLI to version to 3.2.1
OS is Windows 7

platformio.ini is
```
[env:main]
platform = atmelavr
framework = arduino
board = uno
```

Build command stops with error:

> Unknown board ID 'uno'

ivankravets, 2016-12-27:
Could you provide an output of `pio platform list`?

bubnovKelnik, 2017-01-01:
`pio platform list`
: 
`atmelavr ~ Atmel AVR
====================
Atmel AVR 8- and 32-bit MCUs deliver a unique combination of performance, power efficiency and design flexibility. Optimized to speed time to market-and easily adapt to new
 ones-they are based on the industrys most code-efficient architecture for C and assembly programming.

Home: http://platformio.org/platforms/atmelavr
Packages: toolchain-atmelavr, framework-arduinoavr, tool-avrdude
Version: 1.1.0

atmelsam ~ Atmel SAM
====================

Atmel | SMART offers Flash- based ARM products based on the ARM Cortex-M0+, Cortex-M3 and Cortex-M4 architectures, ranging from 8KB to 2MB of Flash including a rich periphe
ral and feature mix.

Home: http://platformio.org/platforms/atmelsam
Packages: framework-arduinosam, tool-avrdude, toolchain-gccarmnoneeabi
Version: 1.1.0`

bubnovKelnik, 2017-01-01:
After updating platformio platform update everything goes well

closed 2017-01-01 by bubnovKelnik