platformio/platformio-core #866: Espressif8266 Error comparing to atmelavr

GeneralLuzi, 2017-01-02:
Hi,
I use the lib DHTLib and recognized an error within the espressif8266-platform
similar main.cpp in both examples:

```
//`
//    FILE: dht33_test.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.01
// PURPOSE: DHT library test sketch for DHT33 && Arduino
//     URL:
//
// Released to the public domain
//
include <dht.h>

dht DHT;

#define DHT33_PIN 9

void setup()
{
    Serial.begin(115200);
    Serial.println("DHT TEST PROGRAM ");
    Serial.print("LIBRARY VERSION: ");
    Serial.println(DHT_LIB_VERSION);
    Serial.println();
    Serial.println("Type,\tstatus,\tHumidity (%),\tTemperature (C)\tTime (us)");
}

void loop()
{
    // READ DATA
    Serial.print("DHT33, \t");

    uint32_t start = micros();
    int chk = DHT.read33(DHT33_PIN);
    uint32_t stop = micros();

    switch (chk)
    {
    case DHTLIB_OK:
        Serial.print("OK,\t");
        break;
    case DHTLIB_ERROR_CHECKSUM:
        Serial.print("Checksum error,\t");
        break;
    case DHTLIB_ERROR_TIMEOUT:
        Serial.print("Time out error,\t");
        break;
    case DHTLIB_ERROR_CONNECT:
        Serial.print("Connect error,\t");
        break;
    case DHTLIB_ERROR_ACK_L:
        Serial.print("Ack Low error,\t");
        break;
    case DHTLIB_ERROR_ACK_H:
        Serial.print("Ack High error,\t");
        break;
    default:
        Serial.print("Unknown error,\t");
        break;
    }
    // DISPLAY DATA
    Serial.print(DHT.humidity, 1);
    Serial.print(",\t");
    Serial.print(DHT.temperature, 1);
    Serial.print(",\t");
    Serial.print(stop - start);
    Serial.println();

    // FOR UNO + DHT33 400ms SEEMS TO BE MINIMUM DELAY BETWEEN SENSOR READS,
    // ADJUST TO YOUR NEEDS
    delay(1000);
}
//
// END OF FILE
//
```
I use the following platform.ini:
```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps = DHTlib
```
and:
```
[env:uno]
platform = atmelavr
board = uno
framework = arduino
lib_deps = DHTlib
```
I get the following error message, when I use the espressif-plattform:

In file included from .piolibdeps\DHTlib_ID1336\dht.h:19:0, from .piolibdeps\DHTlib_ID1336\dht.cpp:41:
.piolibdeps\DHTlib_ID1336\dht.cpp: In member function 'int8_t dht::_readSensor(uint8_t, uint8_t, uint8_t)':
C:\Users\XXX\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:226:63: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in initialization
#define portInputRegister(port)     ((volatile uint32_t*) &GPI)
^
.piolibdeps\DHTlib_ID1336\dht.cpp:121:29: note: in expansion of macro 'portInputRegister'
volatile uint8_t *PIR = portInputRegister(port);
^
*** [.pioenvs\nodemcuv2\lib\DHTlib_ID1336\dht.o] Error 1
When I use the atmelavr platform, no error happens.
Can ypu please check and update the platform.
THX, GL

ivankravets, 2017-01-02:
This library isn't compatible with ESP8266. Fixed in https://github.com/RobTillaart/Arduino/pull/51

Please use another library 

* http://platformio.org/lib/search?query=platform%253Aespressif8266%2520dht*&page=1

closed 2017-01-02 by ivankravets

kiranshashiny, 2017-02-26:
@ivankravets  - The "Simple DHT" library fix - just did not work! 
All I got was "Read DHT11 failed " messages scrolling on the Serial Monitor.

I have a "Node MCU 0.9", and a DHT11 sensor - and it works with Arduino Uno. My Arduino IDE v is 1.6.12

[image]