platformio/platformio-core #868: Support for Nordic nRF52

amq, 2017-01-05:


jrobeson, 2017-01-23:
Should this be a totally separate platform or just use nrf51?

colinblake, 2017-02-15:
I am not completely versed in the difference between the options, but it seems like it could be based on the nrf51.

jrobeson, 2017-02-18:
There are different hardware peripherals available on the nr52, and the next nrf52 (2017 Q3 according to nordic) will even have onboard usb.  I'm not sure at which (or any point) they should be broken out.  Especially when there are board variants of the the nrf51 that allow uploading via virtual usb flash drive (mbed enabled) vs flashing with jtag vs flashing with nrflink.

colinblake, 2017-02-21:
I should add that for my use case, I am interested in loading firmware via
the Arduino boot loader that Sparkfun preflashes on their board.

boneskull, 2017-03-02:
I'm willing to help with this effort if there are others interested (and it sounds like there are).

I just got my paws on a [RedBear BLE Nano v2](https://redbear.cc/product/ble-nano-2.html) which is based on the NRF52832 chip.

As this thing is *brand new*, I haven't had success programming it with anything other than the Arduino IDE--the board [is on GitHub](https://github.com/redbear/nRF5x/), which we can start from.

Is there a quicker way to repackage the above repo into a platform which PlatformIO can use (using Arduino framework)?  My previous attempt(s) at doing this didn't go smoothly.

(mbed doesn't yet recognize the BLE Nano v2, but once it does, the board should also support that framework)

jrobeson, 2017-03-02:
It's a real shame that the development of arduino for the nrf52 is fragmented across 3 different development trees. Redbear has their own, Sparkfun has their own, and then the one by sandeepmistry.  I'm not sure how we should handle that.  And then there's the various recommended flashing methods between them.   I'm not exactly sure how to start :(

boneskull, 2017-03-02:
Hopefully @ivankravets bless us with guidance :smile:

profra, 2017-03-02:
+1 ... I have also RB BLE Nano v2 and it would be nice to have tool for it in Platformio... which I am using for other projects... I made a "port" of Espruino (Puck.js) to RB BLE Nano v2 with tools from here https://github.com/espruino/Espruino

boneskull, 2017-03-02:
@profra cool. this is my first nordic board. I've mostly worked with 8-bit MCUs and ESP8266's.  I wonder the RB Nano2 would run jerryscript+IOT.js. I don't really know what these are capable of.

profra, 2017-03-02:
@boneskull  Thanks... but I am really beginer... having the same experience as you... I had the good guidance of Gordon... check out this thread http://forum.espruino.com/conversations/299650/

PierreKircher, 2017-05-12:
as for the sparkfun breakoutboard and the nordic nrf52 dk both around the 52 832 chip - they are currently in the mail .. im happy to contribute if someone of the core platfrom would be available to give a helping hand - ill do the heavy lifting ( need it for production ) im even willing to pay to get this done

ivankravets, 2017-05-12:
>   need it for production ... im even willing to pay to get this done

@PierreKircher could mail contact@pioplus.com ?

ivankravets, 2017-06-24:
Moved to https://github.com/platformio/platform-nordicnrf52/issues/3

closed 2017-06-24 by ivankravets