platformio/platformio-core #869: Macro contents of __cplusplus are different in Eclipse indexer and "platformio run"

kernkraftzwerg, 2017-01-08:
What kind of issue is this?
- [*] PlatformIO Core.
------------------------------------------------------------------

### Configuration

**Operating system**: Linux (not relevant)

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.2.1

### Description of problem
Eclipse Indexer expands the macro __cplusplus differently to its value in platformio run. This produces many Codan issues if there is used any code depending on the definition of this macro, e.g. thinger.io.

#### Steps to Reproduce

1. Create new project using "platformio init --ide eclipse -b nodemcuv2"

2. In Eclipse open the project and look in project properties -> C/C++ General -> C/C++ Include ... how the macro is defined.

3. Use a code snipped to show the contents of __cplusplus during build:
#define _STR(x) #x
#define _VAL(x) _STR(x)
#pragma message _VAL(__cplusplus)

### Actual Results
In Eclipse: __cplusplus=199711L
In platformio run: __cplusplus=201103L

### Expected Results
Eclipse should have the same macro contents (__cplusplus=201103L)

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
```

### Additional info
The root cause is the Eclipse provider for "CDT Cross GCC Built-in Compiler Settings". This provider calls the compiler using flags for verbose macro expansion and all build flags on an empty source file for retrieving the compiler built-ins. The compiler build flags should be the same for the provider and platformio run, but aren't.
platformio run -v shows these flags: -fno-rtti -fno-exceptions -std=c++11 ...
The provider uses these flags: ${FLAGS}

The flag "-std" is responsible for the contents of __cplusplus. ${FLAGS} is normally set by the Eclipse toolchain, but because the platformio project does not use any toolchain (which is correct) ${FLAGS} is empty. The solution may be either to fill ${FLAGS} with the platformio values in the Eclipse project environment settings or to add the flags in the command line of the Eclipse provider.

sle118, 2017-02-19:
@kernkraftzwerg  thanks for your detailed analysis.  Is there a way to get around this in the short term?

DavidR1953, 2017-03-11:
A short term solution is to go to Project -> Properties -> C/C++ General -> Preprocessor Include Paths, Macros etc.
Select the Providers tab
Click on CDT Cross GCC Built-in Compiler Settings, and in the 'Command to get compiler specs:' box
add -std=c++11 to the command. Something like this:
${HOME}/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++ -std=c++11 ${FLAGS} -E -P -v -dD "${INPUTS}"
Then re-index
This worked for me on Ubuntu 16.10.

hsmptg, 2017-12-17:
Hi
I have a similar problem with the callback of the PubSubClient (MQTT client) in Eclipse:
The compilation is done with success:

_Building .pioenvs\d1_mini\firmware.bin
text	   data	    bss	    dec	    hex	filename
244216	   3444	  30136	 277796	  43d24	.pioenvs\d1_mini\firmware.elf
[SUCCESS] Took 9.69 seconds_

... but I got an annoying error in the editor:

_Invalid arguments '
Candidates are:
PubSubClient & setCallback(?)_

The short term solution of DavidR1953 does not worked for me (at least in Windows)!
Any help?

hsmptg, 2017-12-18:
Hi
There is already an alternative to this "short term solution"?
Because this one solves the "invalid arguments" problem of the indexer but has side effects: now I got errors in print, println and printf of Serial.
Note that all this errors shown by eclipse does not prevent the successful compilation of the code!
Thanks!

ivankravets, 2018-02-08:
Could you reproduce this issue with the latest PIO Core 3.5.1? Please `pio upgrade` and re-init project.

ivankravets, 2018-07-15:
It should be fixed in the latest release.

closed 2018-07-15 by ivankravets

kernkraftzwerg, 2018-07-18:
Let me guess; you removed the settings for "CDT Cross GCC Built-in Compiler Settings" provider? This has really bad side effects to my eclipse projects, now standard include paths form the AVR toolchain are not found anymore. Even eclipse always produces bug reports because this provider crashes on opening of the projects, which is obviously an eclipse bug, but now I know why it bugs me for the last few days ;-) I just reported a bug for the missing include paths.