platformio/platformio-core #870: [Feature Request] Support hardware interaction in native environment

r89m, 2017-01-09:
Hi Ivan,

I'm really enjoying using the unit test features of Platformio to test my code, but I was wondering if there's any possibility of supporting hardware interaction simulation when running in the native environment.

For example, some of the libraries that I write rely on reading data from millis() or digitalRead() and reacting to it. It'd be great if I was able to write tests that simulated these sort of inputs. Additionally it'd be great if I could verify that my code set the correct pin state (eg, using digitalWrite) based on the current conditions.

A while ago I wrote a library [AxxTest](https://github.com/r89m/AxxTest) based on [CxxTest](http://cxxtest.com/) and [FakeIt](https://github.com/eranpeer/FakeIt) which supported a couple of the Arduino hardware functions and allowed such assertions to be made. I've used it on one of two of my libraries such as [MQTTSwitch](https://github.com/r89m/MQTTSwitch) if you'd like to see how it can be used (```tests/MQTTSimpleSwitchTests.h``` gives a decent overview). 

Is this a direction that you'd be interested in PlatformIO taking, or does it open you up to too much maintenance? I'm fairly inexperienced when it comes to C++ but I'd be happy to discuss how the implementation might occur if you're interested.

All the best,

Richard

r89m, 2017-01-10:
I've moved some of my AxxTest work over to Platformio in the following repo if you want to take a look: https://github.com/r89m/Transition

```tests/test_1/test_stepTowardsValue``` makes use of the mocking functionality provided by FakeIt and allows me to make use of the millis() function by providing it with a set of known values.

ivankravets, 2017-01-11:
Seehttps://github.com/platformio/platformio-examples/blob/develop/unit-testing/calculator/platformio.ini#L14

Change [this](https://github.com/r89m/Transition/blob/master/platformio.ini) config to
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:native]
platform=native
build_flags=-std=c++11

[env:uno]
platform = atmelavr
framework = arduino
board = uno
lib_ignore = Arduino
```

It means, that for `uno` build environment the `Arduino` library will be excluded.

Also, we have a lot of different instruments how to control PlatformIO Build System:

1. http://docs.platformio.org/en/latest/projectconf.html#build-flags - define global macro per build environment and place platform specific code under `#ifdef` guard.
2. Customize builder per library: http://docs.platformio.org/en/latest/librarymanager/config.html#build

r89m, 2017-01-12:
Hi Ivan,

Thanks for the tips RE ignoring tests and libraries - that'll definitely be handy when it comes to upload my code to an embedded board.

My actual reason for raising this ticket though was less for support and more to discuss the possibility of adding in hardware virtualisation to the ```native``` test environment.

My current understanding is (please correct me if I'm wrong) that when testing in the native environment it is only possible to test "pure" C++. That is, code with no reference to any Arduino / hardware specific features, such as millis, digitalRead and digitalWrite.

What I propose is that by making use of a mocking framework (I happen to be using FakeIt, but there's plenty of others other there) it becomes possible to test hardware features on the desktop. The tests then become "true" unit tests, in that they can be built, compiled and run all on the host machine with absolutely no requirement for a physical piece of hardware to be attached. This has the benefits of a) being super fast (my PC is faster than an Arduino and there's no need to upload new firmware) and b) portable, in that I can now test hardware features when on the move, in a cafe etc.

My real questions to you I suppose are:
1. Does this interest you?
2. How might I go about adding this to my local native test environment so that I don't have to copy my ```libs/Arduino``` folder to all projects that I want to test

Cheers,

Richard

ivankravets, 2017-01-12:
> Does this interest you?

We support different frameworks and development platforms. I think it will be hard to support "mocking" for them. What do you need, it can be implemented as separate component/library and included/excluded for the specific build environment. The question here is who will create it.

Also, we have a deal here with embedded software that will depend on MCU/Board specification. For example, I'm not sure that FakeIt will run on Arduino Uno.

Honestly, I like your approach with "Arduino" library. However, I recommend to name it as "ArduinoFrameworkFake" or etc. That is a really right solution because for you is OK that `delay()` is implemented as "empty" function. If we talk about "desktop/native", even this `delay()` will have different implementation on Windows/Linux/macOS. That is not an easy work.

> How might I go about adding this to my local native test environment so that I don't have to copy my libs/Arduino folder to all projects that I want to test

You can create a new library named `ArduinoFrameworkFake` and place it somewhere. Then need to add a folder where is located `ArduinoFrameworkFake` to http://docs.platformio.org/en/latest/projectconf.html#lib-extra-dirs

In this case, your `ArduinoFrameworkFake` will be shared between different projects. Finally, you can create custom `ArduinoFrameworkFake` library on GitHub and register in PIO Library Registry.

r89m, 2017-01-12:
First off, thanks for the tip RE ```lib-extra-dirs```, that's definitely the way I'm going to go. Hopefully I'll be able to get it up and running and release it as a library on Platformio.

I think there may be a little misunderstanding though, as there would not be a need for FakeIt to run on real hardware (ie Arduino Uno) or for there to be different implementations for Windows / Linux / macOS but I'll consider this question closed as you have answered my main questions. 

Thanks,

Richard

closed 2017-01-12 by r89m