platformio/platformio-core #876: Need to update tool-stlink

master-q, 2017-01-15:
What kind of issue is this?

- [* ] PlatformIO Core. If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Linux

**PlatformIO Version** (`platformio --version`): 3.2.1

### Description of problem

The st-util tool causes error on STM32F091 Nucleo-64 board. This issue is fixed at latest version of the tool.

#### Steps to Reproduce

1. Using some elf file.
2. Run "~/.platformio/packages/tool-stlink/st-util"
3. Run "arm-none-eabi-gdb foo.elf"
4. Type "tar ext localhost:4242" on gdb
5. Type "load" on gdb

### Actual Results

The gdb console shows:

```
Loading section startup, size 0xc0 lma 0x8000000
Loading section .text, size 0x62ac lma 0x80000c0
Loading section .data, size 0xb8 lma 0x8006370
Error finishing flash operation
```

The st-util console shows:

```
2017-01-15T20:20:47 INFO src/stlink-common.c: Loading device parameters....
2017-01-15T20:20:47 INFO src/stlink-common.c: Device connected is: F09X device, id 0x10006442
2017-01-15T20:20:47 INFO src/stlink-common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
2017-01-15T20:20:47 INFO gdbserver/gdb-server.c: Chip ID is 00000442, Core ID is  0bb11477.
2017-01-15T20:20:47 INFO gdbserver/gdb-server.c: Target voltage is 3253 mV.
2017-01-15T20:20:47 INFO gdbserver/gdb-server.c: Listening at *:4242...
2017-01-15T20:22:18 INFO gdbserver/gdb-server.c: Found 4 hw breakpoint registers
2017-01-15T20:22:18 INFO gdbserver/gdb-server.c: GDB connected.
2017-01-15T20:22:57 INFO src/stlink-common.c: Attempting to write 2048 (0x800) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2017-01-15T20:22:57 INFO src/stlink-common.c: Finished erasing 1 pages of 2048 (0x800) bytes
2017-01-15T20:22:57 INFO src/stlink-common.c: Starting Flash write for VL/F0/F3 core id
2017-01-15T20:22:57 INFO src/stlink-common.c: Successfully loaded flash loader in sram
2017-01-15T20:23:01 ERROR src/stlink-common.c: flash loader run error
2017-01-15T20:23:01 ERROR src/stlink-common.c: run_flash_loader(0x8000000) failed! == -1
```

### Expected Results

The gdb console should show:

```
Loading section startup, size 0xc0 lma 0x8000000
Loading section .text, size 0x62ac lma 0x80000c0
Loading section .data, size 0xb8 lma 0x8006370
Start address 0x80000c0, load size 25636
Transfer rate: 11 KB/sec, 6409 bytes/write.
```

The st-util console should show:

```
[... 88 lines cut ...]
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Attempting to write 2048 (0x800) bytes to stm32 address: 134238208 (0x8005000)
Flash page at addr: 0x08005000 erased
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Finished erasing 1 pages of 2048 (0x800) bytes
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Starting Flash write for VL/F0/F3 core id
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/flash_loader.c: Successfully loaded flash loader in sram

2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Starting verification of write complete
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Flash written and verified! jolly good!
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Attempting to write 2048 (0x800) bytes to stm32 address: 134240256 (0x8005800)
Flash page at addr: 0x08005800 erased
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Finished erasing 1 pages of 2048 (0x800) bytes
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Starting Flash write for VL/F0/F3 core id
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/flash_loader.c: Successfully loaded flash loader in sram

2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Starting verification of write complete
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Flash written and verified! jolly good!
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Attempting to write 2048 (0x800) bytes to stm32 address: 134242304 (0x8006000)
Flash page at addr: 0x08006000 erased
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Finished erasing 1 pages of 2048 (0x800) bytes
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Starting Flash write for VL/F0/F3 core id
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/flash_loader.c: Successfully loaded flash loader in sram

2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Starting verification of write complete
2017-01-15T20:25:49 INFO /home/kiwamu/src/stlink/src/common.c: Flash written and verified! jolly good!
```

ivankravets, 2017-01-15:
Please move issue to https://github.com/platformio/platform-ststm32/issues

closed 2017-01-15 by ivankravets

master-q, 2017-01-16:
Reported. Thanks.

https://github.com/platformio/platform-ststm32/issues/29