platformio/platformio-core #888: Add CI support for custom boards

SRGDamia1, 2017-01-26:
It would be very nice to have the ability to use the continuous integration tools with custom boards!

ivankravets, 2017-01-26:
and why you can't do that?

SRGDamia1, 2017-01-27:
I'm sorry.  Maybe it's just that I can't figure out how.  Every time I run TravisCI over the code on GitHub, it tells me that it can't find the board, even though I have the board spec in a "boards" directory there.  When I run the ci command locally (where I have the board installed in the global directory), it does work without trouble.

SRGDamia1, 2017-01-27:
I also added an issue in the atmelavr repo to add support for our board:  https://github.com/platformio/platform-atmelavr/issues/28

ivankravets, 2017-01-27:
> have the board installed in the global directory), it does work without trouble.

You need to do the same on CI side.

P.S: I'll add your board to our official Atmel AVR development platform and you will use it without any hooks.

closed 2017-01-27 by ivankravets

SRGDamia1, 2017-01-27:
How do you install the board on the CI side?

Thank you so much!

ivankravets, 2017-01-27:
The board has been added to our off repo. Just use it as generic Uno or etc.

P.S: Would be thankful for any support https://pioplus.com/pricing.html