platformio/platformio-core #890: Running PLATFORMIO_REBUILD_PROJECT_INDEX breaks build system

fuho, 2017-01-29:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Ubuntu 16.04 64bit

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.3.0a7

### Description of problem

The problem I am facing now is that after I run `PLATFORMIO_REBUILD_PROJECT_INDEX`  once:
> /usr/bin/cmake --build /home/fuho/Projects/ir_blaster/cmake-build-debug --target PLATFORMIO_REBUILD_PROJECT_INDEX -- -j 8

> The current working directory /home/fuho/Projects/ir_blaster will be used for project.
> You can specify another project directory via
> `platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

> The next files/directories have been created in /home/fuho/Projects/ir_blaster
> platformio.ini - Project Configuration File
> src - Put your source files here
> lib - Put here project specific (private) libraries

> Project has been successfully initialized!
> Useful commands:
> `platformio run` - process/build project from the current directory
> `platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board
> `platformio run --target clean` - clean project (remove compiled files)
> `platformio run --help` - additional information
> Built target PLATFORMIO_REBUILD_PROJECT_INDEX

... the build system breaks (result of `PLATFORMIO_BUILD`):
> /usr/bin/cmake --build /home/fuho/Projects/ir_blaster/cmake-build-debug --target PLATFORMIO_BUILD -- -j 8
> /bin/sh: 1: platformio: not found
> CMakeFiles/PLATFORMIO_BUILD.dir/build.make:57: recipe for target 'CMakeFiles/PLATFORMIO_BUILD' failed
> make[3]: *** [CMakeFiles/PLATFORMIO_BUILD] Error 127
> CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/PLATFORMIO_BUILD.dir/all' failed
> make[2]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/all] Error 2
> CMakeFiles/Makefile2:153: recipe for target 'CMakeFiles/PLATFORMIO_BUILD.dir/rule' failed
> make[1]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/rule] Error 2
> Makefile:144: recipe for target 'PLATFORMIO_BUILD' failed
> make: *** [PLATFORMIO_BUILD] Error 2

.. I assume this is because `CMakeListsPrivate.txt` doesn't have correct location for the platformio command?
> set(ENV{PATH} "/home/fuho/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/lib/jvm/java-8-oracle/bin:/usr/lib/jvm/java-8-oracle/db/bin:/usr/lib/jvm/java-8-oracle/jre/bin")
> set(PLATFORMIO_CMD "platformio")

[... 121 lines cut ...]
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)

add_custom_target(
    PLATFORMIO_UPLOADFS ALL
    COMMAND ${PLATFORMIO_CMD} -f -c clion run --target uploadfs
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)

add_custom_target(
    PLATFORMIO_UPDATE_ALL ALL
    COMMAND ${PLATFORMIO_CMD} -f -c clion update
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)

add_custom_target(
    PLATFORMIO_REBUILD_PROJECT_INDEX ALL
    COMMAND ${PLATFORMIO_CMD} -f -c clion init --ide clion
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)

add_executable(${PROJECT_NAME} ${SRC_LIST})
```

### Additional info

fuho, 2017-01-29:
I think this might be related to [this IntelliJ issue](https://intellij-support.jetbrains.com/hc/en-us/community/posts/208567485-Pycharm-terminal-is-missing-part-of-PATH)..

ivankravets, 2018-02-09:
Could you reproduce this issue with the latest version?

closed 2018-02-13 by ivankravets